intel/media-driver #473: shaders: eliminate trivial differences in regenerated shaders

dvrogozh, 2018-12-17:
Apply #453. Once shaders will be regenerated, you will notice a range of trivial differences comparing to files stored in the repository:
* Tailing whitespaces in vpkrnheader.h
* Windows style line feeds in vpkrnheader.h - please, use Linux line feeds only (`\n`)
* Tailing whitespaces in igvpkrn_g11_icllp.c in the data section
* Not a difference, but nice to be fixed: tailing whitespaces in the license header of vpkrnheader.h

FurongZhang, 2019-01-28:
Since @yangnin has fixed this issue, assign this issue to @yangnin , thanks @yangnin !

dvrogozh, 2019-01-30:
Fixed by https://github.com/intel/media-driver/commit/51463814e3815e7de8379981103709d85e6b83d9

closed 2019-01-30 by dvrogozh