intel/media-driver #483: [MPEG2][Decode] Invalid max frame width/height attribute reporting

dmitryermilov, 2018-12-27:
Driver reports max supported frame width/height as 1920x1920 while it should actually support 2K resolution according to https://github.com/intel/media-driver/blob/master/docs/media_features.md

The bug comes from https://github.com/intel/media-driver/blob/f9b1c19c36ead3cca95295581f0fee8b8b99a137/media_driver/agnostic/common/codec/shared/codec_def_common_encode.h#L53, https://github.com/intel/media-driver/blob/e63034d3b6ab5e44f9f434432fc64990466393d3/media_driver/linux/common/ddi/media_libva_caps.cpp#L475

XinfengZhang, 2018-12-29:
hi @dmitryermilov  could you tell us the codec?

XinfengZhang, 2018-12-29:
1920---> 2048?

dmitryermilov, 2018-12-29:
Hi @XinfengZhang ,

It's MPEG2. 
>1920---> 2048?

Yes.

dmitryermilov, 2019-02-25:
@XinfengZhang , did you have a chance to take a look?

XinfengZhang, 2019-04-18:
@dmitryermilov , sorry , missed it, will file a PR soon

PR #983 (2020-07-06): Correct MAX resolution capability of MPEG2
Bug fix for capability of  MPEG2. fixed #483
commit 6a83e7dc (2020-07-06): Correct MAX resolution capability of MPEG2

Bug fix for capability of  MPEG2. fixed #483
closed 2020-07-06 by intel-mediadev