intel/media-driver #497: Decoder doesn't support linear tiling

illigle, 2019-01-18:
Hello, I tried to display decoded NV12 surface using Vulkan in zero-copy mode
(import PRIME fd into Vulkan Memory),
but since the NV12 surface is not defaulted to linear tiling, the result is not correct.

I tried to allocate custom VA surface in linear tiling, using `drm_intel_bo_alloc`, `drm_intel_bo_gem_export_prime`, `VASurfaceAttribExternalBuffers`.
But the decoded NV12 data is still not in linear layout, but in some tiling or macroblock order.
It seems the decoder doesn't care the tiling of the output surface.

So, is this a bug, or simply I can't create decoded NV12 frame in linear tiling.

Thank you!

illigle, 2019-01-18:
I tried to decode a 1920x1080 video
(Ubuntu 18.10, LibVa 2.2, iHD driver 16.9.git_00f9ae_2018-10-1, Iris 655)
using the the default tiling surface it takes about 1.6 ms per frame,
using the custom linear surface it takes almost the same time.

But if I copy the decoded surface into system memory(using `vaDeriveImage`, `vaMapBuffer`)
when using default surface, it need 0.6 ms to copy one frame,
but when using the linear tiling surface, it only takes 0.1 ms per frame except the result is not in line-by-line order.
(I using _mm256_stream_load_si256 to copy the YUV data)

XinfengZhang, 2019-04-28:
sorry, for late response, it is a bug, we will add swizzling in vaMapBuffer soon, such  as vaGetImage + vaMapBuffer

XinfengZhang, 2020-03-09:
@illigle if you use VA_SURFACE_ATTRIB_MEM_TYPE_VA, the surface will be allocated to linear

dvrogozh, 2020-03-09:
@XinfengZhang : to be honest, it is not clear why using VA_SURFACE_ATTRIB_MEM_TYPE_VA would lead to linear allocation. This seems to be relying on particular driver implementation.

dmitryermilov, 2020-03-09:
> @XinfengZhang : to be honest, it is not clear why using VA_SURFACE_ATTRIB_MEM_TYPE_VA would lead to linear allocation. This seems to be relying on particular driver implementation.

The same is actual to move forward with https://github.com/Intel-Media-SDK/MediaSDK/pull/1987

XinfengZhang, 2020-03-10:
@xhaihao we discussed it before, could you comment it?

xhaihao, 2020-03-17:
If I remembered correctly, VA_SURFACE_ATTRIB_MEM_TYPE_VA means the buffer should be allocated by the backend driver, but user may specify some parameters via VASurfaceAttribExternalBuffers.

XinfengZhang, 2020-05-26:
https://github.com/intel/libva/blob/master/va/va.h#L1346 , need to check the memory type to determine the VASurfaceAttribExternalBufferDescriptor structure. if it is VA_SURFACE_ATTRIB_MEM_TYPE_VA, flags (https://github.com/intel/libva/blob/master/va/va.h#L1409) should include VA_SURFACE_EXTBUF_DESC_ENABLE_TILING