intel/media-driver #498: [MCTF] causes segmentation fault when using hwupload in ffmpeg-qsv

fulinjie, 2019-01-21:
MCTF needs reference frame and will return MFX_ERR_MORE_DATA when the first frame comes. This leads to the vpp failures in sample_vpp when only processing 1 frame. 

In msdk:
/sample_vpp -i /home/linjiefu/Videos/sourceVideo/1920x1080.nv12 -o msdk.yuv -lib hw -sw 1920 -sh 1080 -dw 1920 -dh 1080 -scc nv12 -dcc nv12 -mctf 5 -iopattern d3d_to_sys -async 1 -n 1

will report:
[ERROR], sts=MFX_ERR_UNDEFINED_BEHAVIOR(-16), main, Unexpected error at /home/linjiefu/Documents/git/msdk/samples/sample_vpp/src/sample_vpp.cpp:1105

in FFmpeg qsv:

https://patchwork.ffmpeg.org/patch/11739/
https://patchwork.ffmpeg.org/patch/11740/

ffmpeg -init_hw_device qsv=foo -filter_hw_device foo -v verbose -f rawvideo -pix_fmt nv12 -s:v 352x288 -i /home/linjiefu/Videos/sourceVideo/MOBILE_352x288_30_orig_01.yuv -vf 'hwupload=extra_hw_frames=64,format=qsv,vpp_qsv=mctf=5' -f rawvideo -pix_fmt nv12 -an -vframes 4 -y ./vpp_mctf_4frame.yuv

MCTF works after applying the MCTF patch set.
But it's a work around way to av_usleep(100000) for the release:
[VP]:   EXIT     - Render:3525: Exit VEBOX with CSC output.
[VP]:   NORMAL   - VpHal_RndrRenderVebox:3975: VPOutputPipe = 2, VEFeatureInUse = 1
[VP]:   NORMAL   - RenderSingleStream:840: VEBOX/SFC modified Render Target, skipping further processing.

if drop the av_usleep, segmentation fault will occurs.
It happens when hwupload the second frame and calls MFXVideoCORE_SyncOperation.

However, deinterlace needs reference too, but returns MFX_ERR_NONE when the input contains only one frame.

linjiefu, 2019-03-28:
any comments for this?

onabiull, 2020-05-28:
I've copied this issue to https://github.com/Intel-Media-SDK/MediaSDK/issues/2138. @fulinjie are you ok with closing this one?

fulinjie, 2020-05-28:
sure, thanks,

closed 2020-05-28 by fulinjie