intel/media-driver #499: [APL/BXT] VPP scaling fails with decoded 10-bit (HEVC) data

eero-t, 2019-01-21:
Setup:
* Ubuntu 18.04
* drm-tip kernel v4.20
* MediaSDK and its deps built from Git
* HW: J4205 (18EU APL/BXT)

Test-case:
* 10-bit HEVC decoding, scaling and download

Expected outcome:
* This works on BXT with iHD, as it works fine e.g. on KBL and CFL devices, or when using FFmpeg VA-API with i965 driver

Actual outcome:
* Using scaling causes pipeline to fail, see below for details

MediaSDK sample application; decoding, scaling and downloading P010 data:
```
sample_decode -hw h265 -w 300 -h 300 -p010 -i 1920x1080_10bit_50fps.h265 -o /dev/null
...
Frame number:    0, fps: 0.000, fread_fps: 0.000, fwrite_fps: 0.000
[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), RunDecoding, Unexpected error!! at /opt/builder/source1/media-sdk/samples/sample_decode/src/pipeline_decode.cpp:1961

ERROR: Hardware device was lost or returned unexpected error. Recovering...
pretending that aspect ratio is 1:1
Frame number:    0, fps: 0.000, fread_fps: 0.000, fwrite_fps: 0.000
[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), RunDecoding, Unexpected error!! at /opt/builder/source1/media-sdk/samples/sample_decode/src/pipeline_decode.cpp:1961

ERROR: No input data was consumed since last reset. Quitting to avoid looping forever.
```

Removing the scaling makes the pipeline work.  Or changing to 8-bit input video & NV12 download format.

Same with FFmpeg and VA-API backend:
```
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i 1920x1080_10bit_50fps.h265 -vf scale_vaapi=w=300:h=300,hwdownload,format=p010  -f null -
...
[Parsed_scale_vaapi_0 @ 0x561150ae3980] Failed to start picture processing: 1 (operation failed).
Error while filtering: Input/output error
Failed to inject frame into filter network: Input/output error
Error while processing the decoded data for stream #0:0
Conversion failed!
```
Removing the scaling or switching to 8-bit / NV12 makes pipeline work here too.

FFmpeg QSV backend has some additional issue, as even P010 download seems to be broken on BXT:
```
$ ffmpeg -hwaccel qsv -qsv_device /dev/dri/renderD128 -c:v hevc_qsv -i 1920x1080_10bit_50fps.h265 -vf scale_qsv=w=300:h=300,hwdownload,format=p010 -f null -
...
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (hevc_qsv) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
[NULL @ 0x563fb9408840] missing picture in access unit with size 1
    Last message repeated 1 times
[hevc_qsv @ 0x563fb9419dc0] A decode call did not consume any data: expect more data at input (-10)
[NULL @ 0x563fb9408840] missing picture in access unit with size 1
[hevc_qsv @ 0x563fb9419dc0] A decode call did not consume any data: expect more data at input (-10)
[NULL @ 0x563fb9408840] missing picture in access unit with size 1
[hevc_qsv @ 0x563fb9419dc0] A decode call did not consume any data: expect more data at input (-10)
[NULL @ 0x563fb9408840] missing picture in access unit with size 1
    Last message repeated 2 times
[Parsed_scale_qsv_0 @ 0x563fb94e9340] Error synchronizing the operation: -17
Error while filtering: Unknown error occurred
Failed to inject frame into filter network: Unknown error occurred
Error while processing the decoded data for stream #0:0
Conversion failed!
$ ffmpeg -hwaccel qsv -qsv_device /dev/dri/renderD128 -c:v hevc_qsv -i 1920x1080_10bit_50fps.h265 -vf scale_qsv=w=300:h=300,hwdownload,format=p010 -f null -
...
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (hevc_qsv) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
[NULL @ 0x55e68e82f840] missing picture in access unit with size 1
    Last message repeated 1 times
[hevc_qsv @ 0x55e68e840dc0] A decode call did not consume any data: expect more data at input (-10)
[NULL @ 0x55e68e82f840] missing picture in access unit with size 1
[hevc_qsv @ 0x55e68e840dc0] A decode call did not consume any data: expect more data at input (-10)
[NULL @ 0x55e68e82f840] missing picture in access unit with size 1
[hevc_qsv @ 0x55e68e840dc0] A decode call did not consume any data: expect more data at input (-10)
[NULL @ 0x55e68e82f840] missing picture in access unit with size 1
    Last message repeated 2 times
[AVHWFramesContext @ 0x55e68e868400] Error synchronizing the operation: -17
[hwdownload @ 0x55e68e90ea80] Failed to download frame: -1313558101.
Error while filtering: Unknown error occurred
Failed to inject frame into filter network: Unknown error occurred
Error while processing the decoded data for stream #0:0
Conversion failed!
```

This may be related to #218, but that's from last summer, with Gstreamer and the errors look somewhat different.

KamistixX, 2019-01-27:
Could issue https://github.com/intel/intel-vaapi-driver/issues/398 be related to this one ?

eero-t, 2019-01-28:
> Could issue intel/intel-vaapi-driver#398 be related to this one ?

That issue is only about the output for chroma plane being incorrect, this is about the operation failing completely when scaling is enabled.

FurongZhang, 2019-03-21:
Jister, could you please check this issue?

FurongZhang, 2019-03-21:
@Jister Is it possible for you to check the issue?

eero-t, 2019-06-19:
Tested with latest Git version of media stack and FFmpeg => these cases have been fixed, both with MSDK decode and FFmpeg.

Additionally, both 4K downscaling, and 10-bit  -> 8-bit format conversion in upload case also work with the MSDK decode sample app.

Format conversion doesn't work with FFmpeg QSV or VAAPI backend, and 10-bit 4K doesn't work with VAAPI backend, but as they work for MSDK, those appear to be FFmpeg backend bugs.

closed 2019-06-19 by eero-t