intel/media-driver #500: [VPP] add ARGB and RGB10 surface support when call vaQuerySurfaceAttributes.

zhoujd, 2019-01-23:


FurongZhang, 2019-01-28:
@zhoujd Is it possible for you to let us know the command which can be used to reproduce this issue? Thanks a lot in advance!

zhoujd, 2019-02-01:
1. Get ffmpeg tonemap_vaapi filter code
$ git clone https://gitlab.com/zhoujd/ffmpeg.git
$ git checkout  <branch-name> -b zz/tonemap

2. Run following command line
$ ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i Life_of_Pi_draft_Ultra-HD_HDR.mp4 -vframes 2 -loglevel debug -vf tonemap_vaapi=h2s -f null -

You will got following log -
**[Parsed_tonemap_vaapi_0 @ 0x5584e62e32d0] Hardware does not support output format argb.**
[Parsed_tonemap_vaapi_0 @ 0x5584e62e32d0] Failed to configure output pad on Parsed_tonemap_vaapi_0
Error reinitializing filters!
Failed to inject frame into filter network: Invalid argument
Error while processing the decoded data for stream #0:0

FurongZhang, 2019-02-02:
Thanks for the information. Will follow up this issue ww07 and keep it updated..

zhoujd, 2019-04-11:
may I know some update ?

feiwan1, 2019-06-03:
We meet the same issue is gstreamer-vaapi full test. When we want to convert NV12/I420 into ARGB, the result from vaQuerySurfaceAttributes and filtered by VASurfaceAttribPixelFormat shows that doesn't support ARGB. 

If media-driver support this kind of case, then ARGB should be added into m_vpSurfaceAttr:
https://github.com/intel/media-driver/blob/05ec842840ce76bc665ad3c8905a6450a6a4ca39/media_driver/linux/common/ddi/media_libva_caps.cpp#L69

Our test cmd:
gst-launch-1.0 -vf  videotestsrc num-buffers=1 ! video/x-raw,format=NV12 ! vaapipostproc ! video/x-raw,format=ARGB ! filesink location=test.argb

And, if media-driver do support ARGB format in VP side, then please also check issue #554 , we used different app to convert NV12 to ARGB and BGRA, while the output ARGB and BGRA file is bit match. Guess media-driver used same code for the 2 different format.

zhoujd, 2019-08-27:
Can I know some update regarding this issue ?

zhoujd, 2019-09-11:
Following commit id fixed this issue -
https://github.com/intel/media-driver/commit/f8092716fe1b8d7e2d78f82111e26694f2366739

I am closing this issue.

closed 2019-09-11 by zhoujd