intel/media-driver #501: [ffmpeg-vaapi] add ARGB and RGB10 surface support when call vaCreateImage

zhoujd, 2019-01-23:
ARGB and RGB10 surface is not support in Ddi_Media_CreateImage, need your help to add them to support.

zhoujd, 2019-06-20:
@FurongZhang I met this issue on vaCreateImage, But I think RGB and RGB10 need be supported in vaPutImage and vaGetImage as well.
Can you also consider to add RGB and RGB10 in vaPutImage and vaGetImage functions ?

FurongZhang, 2019-07-04:
I have added 10bit RGB support in vaPutImage and vaGetImage. Code has been merged. Please download the latest code and verify it. Thanks a lot in advance! Please let me know if you are seeing any issues.

zhoujd, 2019-09-11:
After I tested CreateImage/GetImage for ARGB, I got black output in dump file when dump surface into file.

But ARGB in vaDeriveImage path work fine to dump the ARGB surface.

Following code for reference,

    if (ctx->derive_works && dst->format == hwfc->sw_format &&
        ((flags & AV_HWFRAME_MAP_DIRECT) || !(flags & AV_HWFRAME_MAP_READ))) {
        vas = **vaDeriveImage**(hwctx->display, surface_id, &map->image);
        if (vas != VA_STATUS_SUCCESS) {
            av_log(hwfc, AV_LOG_ERROR, "Failed to derive image from "
                   "surface %#x: %d (%s).\n",
                   surface_id, vas, vaErrorStr(vas));
            err = AVERROR(EIO);
            goto fail;
        }
        if (map->image.format.fourcc != image_format->fourcc) {
            av_log(hwfc, AV_LOG_ERROR, "Derive image of surface %#x "
                   "is in wrong format: expected %#08x, got %#08x.\n",
                   surface_id, image_format->fourcc, map->image.format.fourcc);
            err = AVERROR(EIO);
            goto fail;
        }
        map->flags |= AV_HWFRAME_MAP_DIRECT;
    } else {
        vas = **vaCreateImage**(hwctx->display, image_format,
                            hwfc->width, hwfc->height, &map->image);
        if (vas != VA_STATUS_SUCCESS) {
            av_log(hwfc, AV_LOG_ERROR, "Failed to create image for "
                   "surface %#x: %d (%s).\n",
                   surface_id, vas, vaErrorStr(vas));
            err = AVERROR(EIO);
            goto fail;
        }
        if (!(flags & AV_HWFRAME_MAP_OVERWRITE)) {
            vas = **vaGetImage**(hwctx->display, surface_id, 0, 0,
                             hwfc->width, hwfc->height, map->image.image_id);
            if (vas != VA_STATUS_SUCCESS) {
                av_log(hwfc, AV_LOG_ERROR, "Failed to read image from "
                       "surface %#x: %d (%s).\n",
                       surface_id, vas, vaErrorStr(vas));
                err = AVERROR(EIO);
                goto fail;
            }
        }
    }

zhoujd, 2019-09-18:
For ARGB, vaDeriveImage can work.

zhoujd, 2019-10-16:
For RGB10 createimage/vagetimage is working well, I am closing this issue.

closed 2019-10-16 by zhoujd