intel/media-driver #502: [ffmpeg-vaapi][jpeg] Media_Format_IMC3 is not supported in DdiVp_GetColorSpaceFromMediaFormat

zhoujd, 2019-01-23:
when run loadjpeg and jpeg have IMC3 surface, put image will failed because Media_Format_IMC3 is not supported in DdiVp_GetColorSpaceFromMediaFormat.

$ loadjpeg 61a54ilgerc1i1bqyo6rna4z.jpg 
WARNING:Height need to be a multiple of 16 (current height is 1240)
WARNING:Width need to be a multiple of 16 (current Width is 964)
Decoding JPEG image 964x1240...
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /opt/zach/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_4
libva info: va_openDriver() returns 0
[VP]:   CRITICAL - DdiVp_GetColorSpaceFromMediaFormat:1740: Get ColorSpace from media format: unknown format.

press any key to exit23

Decoding finished in 77617 ticks

FurongZhang, 2019-01-28:
@zhoujd  Is it possible for you to share the command which can be used to reproduce this issue? If there is no command, we can not know how to reproduce this issue. Thanks a lot in advance!

zhoujd, 2019-01-29:
@FurongZhang  the command line is -
$ loadjpeg 61a54ilgerc1i1bqyo6rna4z.jpg

I can send the file 61a54ilgerc1i1bqyo6rna4z.jpg in email.

libva-utils loadjpeg can reproduce it.

zhoujd, 2019-04-03:
I am closing this issue.

PR #530 (2019-04-24): [VP]Add IMC3 in DdiVp_GetColorSpaceFromMediaFormat
Fix #502
closed 2019-04-03 by zhoujd