intel/media-driver #504: build in Fedora: can't find __user in i915 header file

chivakker, 2019-01-25:
When compiling in Fedora 29 with gcc (GCC) 8.2.1 20181215 (Red Hat 8.2.1-6)

It seems definition of __user is missing, since this is a kernel only descriptor, can a long term solution with no i915 headers be considered?

```
In file included from <redacted>/include/xf86drm.h:40,
                 from <redacted>/media-driver/media_driver/linux/common/ddi/media_libva_common.h:33,
                 from <redacted>/media-driver/media_driver/linux/common/cm/media_libva_cm.h:30,
                 from <redacted>/media-driver/media_driver/linux/common/cm/cm_def_os.h:41,
                 from <redacted>/media-driver/media_driver/agnostic/common/cm/cm_def.h:30,
                 from <redacted>/media-driver/media_driver/agnostic/common/cm/cm_array.h:30,
                 from <redacted>/media-driver/media_driver/agnostic/common/cm/cm_array.cpp:27:
<redacted>//media-driver/media_driver/linux/common/os/i915/include/uapi/drm.h:144:7: error: expected ‘;’ at end of member declaration
  char __user *name;   /**< Name of driver */
```

chivakker, 2019-01-25:
igt-gpu-tools doesn't carry the __user identifier in those headers.  Using those headers or better their method should fix this problem.

chivakker, 2019-01-25:
from igt-gpu-tools README file

**include/drm-uapi**

Imported DRM uapi headers from airlied's drm-next branch.
These should be updated all together by executing "make
headers_install" from that branch of the kernel and then
copying the resulting ./usr/include/drm/*.h in and committing
with a note of which commit on airlied's branch was used to
generate them.

uartie, 2019-01-25:
FWIW, I can't reproduce this error.  I compiled the latest master libva, gmmlib and media-driver in a clean Fedora 29 Docker container with system installed `libdrm-devel 2.4.96` and `g++ (GCC) 8.2.1 20181215 (Red Hat 8.2.1-6)`.

Perhaps you have a polluted environment?

uartie, 2019-01-25:
Also, according to our CI, we haven't had a single build failure on Fedora 29 for master media-driver since Dec 25th (unrelated).  Latest full master build at https://media-ci.ostc.intel.com/build/229843 passed.

XinfengZhang, 2019-01-25:
@chivakker yes, maybe we can pick the headers after "make
headers_install". will file a PR. but I still dont understand why you meet the issues. the define should exist there.

dvrogozh, 2019-01-28:
@chivakker: can you, please, try #505 to see if your build on Fedora succeeds?