intel/media-driver #507: Failed to initialize on CLF (i9-9900K)

wakuken, 2019-01-28:
I built up two Cofee Lake Refresh-S computers (Core i9-9900K and i7-9700K / Intel Z390 chipset MB).
First, I installed drivers on Ubuntu 16.04 LTS + Kernel v4.14.16 by executing build_msdk_ubuntu.sh but VAAPI does not work on my both computers.

So I upgraded Ubuntu to 18.04 LTS, changed to Kernel v4.18.0 (sudo apt install linux-generic-hwe-18.04), and upgraded graphics driver (oibaf/graphics-drivers) but it does not work also.
Ofcourse i915.alpha_support=1 is added into the boot option.
On my another Core i7-8650U (KaybeLake) NUC, it works well by using iHD.

Anyone can help me? Thank you.

```
$ vainfo
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_4
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit

$ inxi -F
System:    Host: xxxx Kernel: 4.18.0-13-generic x86_64 bits: 64 Console: tty 2
           Distro: Ubuntu 18.04.1 LTS
Machine:   Device: desktop Mobo: ASUSTeK model: ROG STRIX Z390-F GAMING v: Rev 1.xx serial: N/A
           UEFI: American Megatrends v: 0506 date: 09/20/2018
CPU:       8 core Intel Core i9-9900K (-MT-MCP-) cache: 16384 KB
           clock speeds: max: 5000 MHz 1: 950 MHz 2: 3820 MHz 3: 4378 MHz 4: 4391 MHz 5: 4389 MHz
           6: 3821 MHz 7: 3934 MHz 8: 3875 MHz 9: 4361 MHz 10: 4381 MHz 11: 4394 MHz 12: 3852 MHz
           13: 4390 MHz 14: 4387 MHz 15: 3827 MHz 16: 3825 MHz
Graphics:  Card: Intel Device 3e98
           Display Server: N/A drivers: modesetting (unloaded: fbdev,vesa)
           tty size: 111x24 Advanced Data: N/A out of X
Audio:     Card Intel Device a348 driver: snd_hda_intel Sound: ALSA v: k4.18.0-13-generic
Network:   Card: Intel Ethernet Connection (7) I219-V driver: e1000e
           IF: eno1 state: up speed: 100 Mbps duplex: full mac: 04:92:26:c2:52:33
Drives:    HDD Total Size: 750.2GB (4.3% used)
           ID-1: /dev/nvme0n1 model: Samsung_SSD_970_EVO_250GB size: 250.1GB
           ID-2: /dev/sda model: SanDisk_SDSSDH35 size: 500.1GB
Partition: ID-1: / size: 214G used: 16G (8%) fs: ext4 dev: /dev/nvme0n1p2
           ID-2: swap-1 size: 17.02GB used: 0.00GB (0%) fs: swap dev: /dev/nvme0n1p3
RAID:      No RAID devices: /proc/mdstat, md_mod kernel module present
Sensors:   System Temperatures: cpu: 27.8C mobo: N/A
           Fan Speeds (in rpm): cpu: 0
Info:      Processes: 349 Uptime: 3 days Memory: 1492.5/15880.1MB Init: systemd runlevel: 5
           Client: Shell (bash) inxi: 2.3.56

$ lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:3e30] (rev 0a)
00:01.0 PCI bridge [0604]: Intel Corporation Skylake PCIe Controller (x16) [8086:1901] (rev 0a)
00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:3e98]
00:14.0 USB controller [0c03]: Intel Corporation Device [8086:a36d] (rev 10)
00:14.2 RAM memory [0500]: Intel Corporation Device [8086:a36f] (rev 10)
00:16.0 Communication controller [0780]: Intel Corporation Device [8086:a360] (rev 10)
00:17.0 SATA controller [0106]: Intel Corporation Device [8086:a352] (rev 10)
00:1b.0 PCI bridge [0604]: Intel Corporation Device [8086:a340] (rev f0)
00:1c.0 PCI bridge [0604]: Intel Corporation Device [8086:a338] (rev f0)
00:1d.0 PCI bridge [0604]: Intel Corporation Device [8086:a330] (rev f0)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:a305] (rev 10)
00:1f.3 Audio device [0403]: Intel Corporation Device [8086:a348] (rev 10)
00:1f.4 SMBus [0c05]: Intel Corporation Device [8086:a323] (rev 10)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Device [8086:a324] (rev 10)
00:1f.6 Ethernet controller [0200]: Intel Corporation Ethernet Connection (7) I219-V [8086:15bc] (rev 10)
01:00.0 Multimedia video controller [0400]: Blackmagic Design DeckLink 8K Pro [bdbd:a14b]
04:00.0 Non-Volatile memory controller [0108]: Samsung Electronics Co Ltd Device [144d:a808]

$ dmesg | grep drm
[    1.390896] fb: switching to inteldrmfb from EFI VGA
[    1.390951] [drm] Replacing VGA console driver
[    1.391916] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    1.391916] [drm] Driver supports precise vblank timestamp query.
[    1.392575] [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
[    1.665152] [drm] failed to retrieve link info, disabling eDP
[    1.667409] [drm] Initialized i915 1.6.0 20180514 for 0000:00:02.0 on minor 0
[    1.706461] fbcon: inteldrmfb (fb0) is primary device
[    1.706512] i915 0000:00:02.0: fb0: inteldrmfb frame buffer device
```

dvrogozh, 2019-02-01:
@wakuken: can you, please, check whether the above commit fixes the issue for you? Please, start with the kernel you were using.

wakuken, 2019-02-01:
Hello @dvrogozh

Thank you for your support!

After applying your patch, I confirmed to decode H264/H265 4K stream in my live streaming app.

It succeeded in decoding the stream, but the **decoding speed is too slow**.

KBL(Core i7-7700): about 100 fps.
This CFL(Core i9-9900): about 40 fps.

My app uses system memory and set AsyncDepth to 1. When it uses VAAPI memory, the decoding in the CFL becomes as fast as in the KBL (200 fps!). But the memory copy to the system memory is too slow.

The kernel is still 4.18 because blackmagic's decklink driver does not support 4.20 currently.

Does the newest kernel improve the decoding speed? Or is there another bug somewhere?

Thank you.

wakuken, 2019-02-01:
This situation can be also reproduced with MediaSDK's sample decoder.
```
In the CFL (i9-9900K):
$ ./sample_decode h265 -i ~/8sec.es -async 1
plugin_loader.h :186 [INFO] Plugin was loaded from GUID: { 0x33, 0xa6, 0x1c, 0x0b, 0x4c, 0x27, 0x45, 0x4c, 0xa8, 0xd8, 0x5d, 0xde, 0x75, 0x7c, 0x6f, 0x8e } (Intel (R) Media SDK HW plugin for HEVC DECODE)
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_4
libva info: va_openDriver() returns 0
Decoding Sample Version 8.3.26.
Input video	HEVC
Output format	NV12
Input:
  Resolution	3840x2160
  Crop X,Y,W,H	0,0,0,0
Output:
  Resolution	3840x2160
Frame rate	59.94
Memory type		system
MediaSDK impl		hw
MediaSDK version	1.28
Decoding started
Frame number:  480, fps: 43.071, fread_fps: 0.000, fwrite_fps: 0.000
Decoding finished
```

```
In the KBL(i7-7700):
$ ./sample_decode h265 -i ~/8sec.es -async 1
plugin_loader.h :186 [INFO] Plugin was loaded from GUID: { 0x33, 0xa6, 0x1c, 0x0b, 0x4c, 0x27, 0x45, 0x4c, 0xa8, 0xd8, 0x5d, 0xde, 0x75, 0x7c, 0x6f, 0x8e } (Intel (R) Media SDK HW plugin for HEVC DECODE)
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_4
libva info: va_openDriver() returns 0
Decoding Sample Version 8.3.26.
Input video	HEVC
Output format	NV12
Input:
  Resolution	3840x2160
  Crop X,Y,W,H	0,0,0,0
Output:
  Resolution	3840x2160
Frame rate	59.94
Memory type		system
MediaSDK impl		hw
MediaSDK version	1.28
Decoding started
Frame number:  480, fps: 108.622, fread_fps: 0.000, fwrite_fps: 0.000
Decoding finished
```

closed 2019-02-02 by Sherry-Lin

dvrogozh, 2019-02-02:
@wakuken : Please, open separate issue to track performance problem. Which application you were using to initially reproduce CFL slow memory copy? Was it mediasdk based?

I believe mediasdk also miss the 0x3e98 device ID. Guys will add it to mediasdk in the beginning of the next week and I will let you know. Let's recheck then again the perf issue....

wakuken, 2019-02-04:
Hello @dvrogozh
Thank you for your help.
My original application is based on MediaSDK. The decoding scheme is almost the same as sample_decode (Set AsyncDepth to 1 and system memory).
MediaSDK also checks the device ID, doesn't it?
I will check again after the fix.

PR #517 (2019-02-02): [Encode] Add some missing device IDs.
Fixes: #507
commit ddf5b775 (2019-02-02): [Encode] Add some missing device IDs.
Fixes: #507
commit 1d28de1a (2019-02-03): [Encode] Add some missing device IDs.
Fixes: #507