intel/media-driver #515: Do we really support some device IDs: eliminate triple list of device IDs

dvrogozh, 2019-02-01:
Right now media driver uses 3 lists of device IDs:
1. First list is located in gmmlib: https://github.com/intel/gmmlib/blob/master/Source/inc/common/igfxfmid.h
2. Second list located in i915 backend of media-driver: https://github.com/intel/media-driver/blob/master/media_driver/linux/common/os/i915/include/intel_chipset.h
3. Third list located in a bunch of files:
3.1. https://github.com/intel/media-driver/blob/master/media_driver/linux/gen8/ddi/media_sysinfo_g8.cpp
3.2. https://github.com/intel/media-driver/blob/master/media_driver/linux/gen9/ddi/media_sysinfo_g9.cpp
3.3. https://github.com/intel/media-driver/blob/master/media_driver/linux/gen10/ddi/media_sysinfo_g10.cpp
3.4. https://github.com/intel/media-driver/blob/master/media_driver/linux/gen11/ddi/media_sysinfo_g11.cpp
In the following form:
```
static bool icllpDevice8a51 = DeviceInfoFactory<GfxDeviceInfo>::
    RegisterDevice(0x8a51, &icllpGt2Info);
```

As below examples show, these 3 lists are in clear dis-sync. It is quite difficult to understand which device IDs are supported by media-driver in reality. Could we, please, eliminate duplicated lists and use a single one? I strongly suggest to end up with the as straightforward single file holding device IDs list as possible. Consider examples from mesa and kernel:
* https://cgit.freedesktop.org/mesa/mesa/tree/include/pci_ids/i965_pci_ids.h
* https://cgit.freedesktop.org/drm/drm/tree/include/drm/i915_pciids.h

Examples:
1. Consider CFL 0x3E91 device ID and grep for it in media-driver and gmmlib (which is used by media-driver):
```
[dvrogozh@dvrscl github]$ fgrep -rsn 0x3E91 gmmlib/
gmmlib/Source/inc/common/igfxfmid.h:726:#define ICFL_GT2_S42_DT_DEVICE_F0_ID            0x3E91
gmmlib/install/include/igdgmm/inc/common/igfxfmid.h:726:#define ICFL_GT2_S42_DT_DEVICE_F0_ID            0x3E91
[dvrogozh@dvrscl github]$
[dvrogozh@dvrscl github]$
[dvrogozh@dvrscl github]$ fgrep -rsn 0x3E91 media-driver/
media-driver/media_driver/linux/common/os/i915/include/intel_chipset.h:223:#define PCI_CHIP_COFFEELAKE_S_GT2_1     0x3E91
[dvrogozh@dvrscl github]$
[dvrogozh@dvrscl github]$
[dvrogozh@dvrscl github]$ fgrep -rsn 0x3e91 media-driver/ | grep RegisterDevice
media-driver/media_driver/linux/gen9/ddi/media_sysinfo_g9.cpp:814:    RegisterDevice(0x3e91, &cflGt1Info);
[dvrogozh@dvrscl github]$
```

Thus, gmmlib and i915 backend in media-driver classifies the device as GT2, but media-driver registers it as GT1. Does media driver properly support this device?

2. Consider CFL device ID 0x3EA1:
```
[dvrogozh@dvrscl github]$ fgrep -rsn 0x3EA1 gmmlib/
gmmlib/Source/inc/common/igfxfmid.h:736:#define ICFL_GT1_U41F_U2F1F_ULT_DEVICE_F0_ID    0x3EA1
gmmlib/install/include/igdgmm/inc/common/igfxfmid.h:736:#define ICFL_GT1_U41F_U2F1F_ULT_DEVICE_F0_ID    0x3EA1
[dvrogozh@dvrscl github]$
[dvrogozh@dvrscl github]$
[dvrogozh@dvrscl github]$ fgrep -rsn 0x3EA1 media-driver/
[dvrogozh@dvrscl github]$
[dvrogozh@dvrscl github]$
[dvrogozh@dvrscl github]$ fgrep -rsn 0x3ea1 media-driver/ | grep RegisterDevice
media-driver/media_driver/linux/gen9/ddi/media_sysinfo_g9.cpp:835:    RegisterDevice(0x3ea1, &cflGt1Info);
[dvrogozh@dvrscl github]$
```

Thus, i915 backend of media driver knows nothing about 0x3EA1. Does media-driver really supports this device?

3. Consider CFL device ID 0x3E99:
```
[dvrogozh@dvrscl github]$ fgrep -rsn 0x3E99 gmmlib/
gmmlib/Source/inc/common/igfxfmid.h:733:#define ICFL_GT1_S6_S4_S2_F1F_DT_DEVICE_F0_ID   0x3E99
gmmlib/install/include/igdgmm/inc/common/igfxfmid.h:733:#define ICFL_GT1_S6_S4_S2_F1F_DT_DEVICE_F0_ID   0x3E99
[dvrogozh@dvrscl github]$
[dvrogozh@dvrscl github]$
[dvrogozh@dvrscl github]$ fgrep -rsn 0x3E99 media-driver/
[dvrogozh@dvrscl github]$
[dvrogozh@dvrscl github]$
[dvrogozh@dvrscl github]$ fgrep -rsn 0x3E99 media-driver/ | grep RegisterDevice
[dvrogozh@dvrscl github]$
```

Thus, only gmmlib knows about this device. Does media driver support it?

wangyan42164, 2019-02-12:
1. I do some initial check, the macro definition of intel_chipset.h seems not be used in media-driver. So it may be able to be removed.
2. media_sysinfo_xxx.cpp can use the macro definition of igfxfmid.h of gmmlib instead of the hard-coded id number.

XinfengZhang, 2020-03-09:
@dvrogozh , it could be closed, right?

closed 2020-05-14 by XinfengZhang