intel/media-driver #519: Performance issue on CFL (i9-9900K)

wakuken, 2019-02-04:
This is separated from #507.

I found a performance issue on my Core i9-9900K Cofee Lake Refresh-S.
In my application based on MediaSDK, the **decoding speed is too slower** than previous CPU.

KBL(Core i7-7700): about 100 fps.
This CFL(Core i9-9900): about 40 fps.

My app uses system memory and set AsyncDepth to 1.
As a trial, I added "{ 0x3E98, MFX_HW_CFL, MFX_GT2 }, " into listLegalDevIDs[] in libmfx_core_vaapi.cpp, but it produced no effect.

This situation can be also reproduced with MediaSDK's sample decoder.

```
In the CFL (i9-9900K):
$ ./sample_decode h265 -i ~/8sec.es -async 1
plugin_loader.h :186 [INFO] Plugin was loaded from GUID: { 0x33, 0xa6, 0x1c, 0x0b, 0x4c, 0x27, 0x45, 0x4c, 0xa8, 0xd8, 0x5d, 0xde, 0x75, 0x7c, 0x6f, 0x8e } (Intel (R) Media SDK HW plugin for HEVC DECODE)
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_4
libva info: va_openDriver() returns 0
Decoding Sample Version 8.3.26.
Input video	HEVC
Output format	NV12
Input:
  Resolution	3840x2160
  Crop X,Y,W,H	0,0,0,0
Output:
  Resolution	3840x2160
Frame rate	59.94
Memory type		system
MediaSDK impl		hw
MediaSDK version	1.28
Decoding started
Frame number:  480, fps: 43.071, fread_fps: 0.000, fwrite_fps: 0.000
Decoding finished
```

```
In the KBL(i7-7700):
$ ./sample_decode h265 -i ~/8sec.es -async 1
plugin_loader.h :186 [INFO] Plugin was loaded from GUID: { 0x33, 0xa6, 0x1c, 0x0b, 0x4c, 0x27, 0x45, 0x4c, 0xa8, 0xd8, 0x5d, 0xde, 0x75, 0x7c, 0x6f, 0x8e } (Intel (R) Media SDK HW plugin for HEVC DECODE)
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_4
libva info: va_openDriver() returns 0
Decoding Sample Version 8.3.26.
Input video	HEVC
Output format	NV12
Input:
  Resolution	3840x2160
  Crop X,Y,W,H	0,0,0,0
Output:
  Resolution	3840x2160
Frame rate	59.94
Memory type		system
MediaSDK impl		hw
MediaSDK version	1.28
Decoding started
Frame number:  480, fps: 108.622, fread_fps: 0.000, fwrite_fps: 0.000
Decoding finished
```

wakuken, 2019-02-05:
I updated the kernel to v4.20.6, but it also had no effect on the performance.

dvrogozh, 2019-02-05:
Kernel should not matter much here. At least 4.18 vs. 4.20. That's something else, I think.

Can you, please, double check that you see the issue on the latest mediasdk from master? Make sure that you have this commit: https://github.com/Intel-Media-SDK/MediaSDK/commit/611cb30a6722c420cfd17ce329f975b2fe79f3b0 which propagates your device ID to mediasdk.

wakuken, 2019-02-06:
Hi @dvrogozh ,
Thank you for your advice.
I updated MediaSDK from master, but there seems to be no change in performance.

dvrogozh, 2019-02-06:
@wakuken : ok, thank you. Well, at least we now know that platform should be fully propagated now. Let me think a little bit, please.

wakuken, 2019-02-06:
Hi @dvrogozh ,

I noticed that the current GPU frequency is always set to 350 Mhz on my CFL, so I changed the minimum frequency.

# cat /sys/bus/pci/devices/0000\:00\:02.0/drm/card0/gt_max_freq_mhz > /sys/bus/pci/devices/0000\:00\:02.0/drm/card0/gt_min_freq_mhz

Then, the decoding speed has been improved greatly! It has increased to 100fps!
Is this solution correct?

In my other machine (KBL NUC), the GPU's frequency is automatically changed corresponding to the load.

dvrogozh, 2019-02-06:
Wow. Of course, frequency has direct effect over the speed. However, this is not a correct solution - it just demonstrates that something goes wrong in i915 driver when it scales frequency. You can use it as a workaround, but there is a bug somewhere, most likely in the kernel. So, you actually greatly narrowed down the problem, thank you:).

We need to return to the kernel and essentially follow this: https://01.org/linuxgraphics/documentation/how-report-bugs

Major part of experiment would be:
1. Try to reproduce the issue on drm-tip kernel: `git clone https://cgit.freedesktop.org/drm-tip`
2. When you reproduce the issue boot the kernel with `drm.debug=0x1e log_buf_len=1M` and provide _full_ dmesg log from the boot time (this really can be important in our case, we need to check boot stage)
3. Provide information here along with full kernel cmdline command

Could you, please, do that?

If you will fail to reproduce on drm-tip - this also would be a result since some probably patch got merged. By the way, you can first catch dmesg for your kernel (letter kernel - the better) with `drm.debug=0x1e log_buf_len=1M`

wakuken, 2019-02-08:
Hi @dvrogozh ,

I'm sorry for the late reply.
First of all, the decoding speed is improved on the drm-tip kernel!
It seems that the frequency has changed flexibly. (I checked /sys/bus/pci/devices/0000\:00\:02.0/drm/card0/gt_cur_freq_mhz.)

As you can see, the problem has been fixed between 4.20.6 and 5.0.0rc5+(drm-tip kernel).

Please check the following dmesg's logs.
[boot_dmesg_4.20.6.txt](https://github.com/intel/media-driver/files/2845332/boot_dmesg_4.20.6.txt)
[boot_dmesg_5.0.0-rc5+.txt](https://github.com/intel/media-driver/files/2845333/boot_dmesg_5.0.0-rc5%2B.txt)

Xiaogangli-intel, 2019-05-16:
Thank you @dvrogozh for helping.
Hi @wakuken,
     So the issue has been fixed, right? Can we close this now?

wakuken, 2019-05-17:
Hi @Xiaogangli-intel,

I tried Kernel v5.0.13 (https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.0.13/) on Ubuntu 18.04 LTS.
Unfortunately the kernel seems to have the same problem.

XinfengZhang, 2020-07-06:
@wakuken  it will take time to merge the code from drm-tips to DRM , then to kernel master, so v5.0.13 of drm-tips works , not mean v5.0.13 work. I suppose it already in DRM and kernel master now.  you also could file issue with DRM.  I will close it, if you have further concern, please let us know, thanks

closed 2020-07-06 by XinfengZhang