intel/media-driver #520: [Intel GVT-g] Fail to create context with mediated pass-through GPU

snir911, 2019-02-07:
I have a VM with virtual mediated pass-through GPU (Intel GVT-g), when i'm trying to encode h264/h265 from the guest using libva-utils encoding utilities it fails with the following error:
"setup_encode:caCreateContext (1306) failed,exit" (returned value 1).

Additional info:
-Fedora 29 both on host and guest, work fine on host.
-It works well with intel-vaapi-driver(i965) (the issue is only with intel-media-driver(iHD))
-vainfo does show these encoding capabilities
-SKL

XinfengZhang, 2019-02-17:
sorry for late response, are you using hevcenc or avcenc?

snir911, 2019-02-17:
I tried both hevcencode and h264encode both failed with same error (but on hevcencode it's line 2310)

XinfengZhang, 2019-02-17:
only CreateContext error? vaInitialize return VA_STATUS_SUCCESS

snir911, 2019-02-18:
I did really not look into the code, I got only this error when i tried it. But I think it did pass the initialization successfully if it got to the encoding phase

TerrenceXu, 2019-02-19:
@snir911 , I am from Intel GVT-g team can you file this issue in below link with the detail error log? Thanks!
https://github.com/intel/gvt-linux/issues

closed 2019-08-29 by XinfengZhang