intel/media-driver #532: No packed 24 RGB  image format support

HeJunyan, 2019-02-20:
Need to support packed 24 RGB image format as input.

FurongZhang, 2019-02-26:
May I know the reason why packed 24 RGB image format is needed? For what kind of usage? Thanks a lot in advance!

HeJunyan, 2019-03-01:
I know this is a rarely used format but the request is we need to support:
gst-launch-1.0 videotestsrc ! video/x-raw, format=RGB, framerate=30/1, width=640, height=360 ! vaapipostproc ! glimagesink

here format=RGB is packed 24 bits RGB

dnagasun, 2019-03-13:
> May I know the reason why packed 24 RGB image format is needed? For what kind of usage? Thanks a lot in advance!

TSD's customer are using this format for camera support. So there is need to enable this for IOTG product.

FurongZhang, 2019-04-11:
https://github.com/intel/media-driver/pull/594

FurongZhang, 2019-04-15:
PR has been merged.

FurongZhang, 2019-04-21:
@HeJunyan  This fix has been merged into master branch. Is it possible for you to close this issue? Thanks a lot in advance!

HeJunyan, 2019-04-22:
Verified and OK

closed 2019-04-22 by HeJunyan