intel/media-driver #533: Need to add RGB16(RGB565) image format support

HeJunyan, 2019-02-20:
No RGB16(RGB565) image format support now. The RGB565 or BGR565 image format should be added.

FurongZhang, 2019-02-26:
Will check it and get back.

FurongZhang, 2019-02-26:
May I know what the usage is for RGB16(565)? Now, the trending is 8bit, 10bit. May I know if there is someone who is using RGB565? Thanks a lot in advance!

HeJunyan, 2019-02-26:
Some requirement says:

 gst-launch-1.0 filesrc location= /media/Raw_Videos/Prog_Planar_RGB16_320x240_5frames.yuv ! videoparse width=320 height=240 format=rgb16 ! vaapipostproc format=nv12 ! glimagesink

Here, rgb16 is the RGB565 format.

eero-t, 2019-03-01:
> May I know what the usage is for RGB16(565)? Now, the trending is 8bit, 10bit.

Trend for 8 & 10-bit data is mainly for human eyes.  Neural networks related software typically requires RGB input,  YUV is rarely used there.  They process data with low number of bits, both for training the network (with huge amount of input video) and when the trained network processes data.  Don't know whether anything specifically uses 565 though.

dnagasun, 2019-03-13:
TSD's customer are using this format for camera support. So there is need to enable this for IOTG product.

dnagasun, 2019-03-13:
> May I know what the usage is for RGB16(565)? Now, the trending is 8bit, 10bit. May I know if there is someone who is using RGB565? Thanks a lot in advance!

TSD's customer are using this format for camera support. Specifically for Mondelo. So there is need to enable this for IOTG product. Thanks in advance for your support.

FurongZhang, 2019-03-14:
Sure. Then, we will support that. Thanks a lot for the information.

PR #573 (2019-04-11): [VP]Add RGB565 support in vaCreateImage
Fix #533
closed 2019-04-11 by intel-mediadev