intel/media-driver #538: Need to add ARGB-10LE image format support

HeJunyan, 2019-02-26:
Need to support 10bits ARGB format.  The display may support 10bits ARGB input.
E.G.
gst-launch-1.0 videotestsrc ! capsfilter caps=video/x-raw,format=P010_10LE ! vaapipostproc format=argb-10le ! glimagesink

need 10bits ARGB image as input to display.

FurongZhang, 2019-02-26:
I have added 10bit RGB in Libva. Currently, the pull request is being reviewing. Driver side, will also submit a patch.

xhaihao, 2019-03-21:
@FurongZhang @XinfengZhang what is the fourcc code for 10bit RGB?

FurongZhang, 2019-03-21:
@xhaihao  I added a PR in Libva. And I am fixing the RGB10 format issues, will merge into master after I finish the fixes.

FurongZhang, 2019-03-21:
@xhaihao  I have already added you into the link. Is it possible for you to help to review that? Thanks a lot in advance!

FurongZhang, 2019-03-21:
@xhaihao Your merge about ARGB10 fixed your issue?

xhaihao, 2019-03-22:
@FurongZhang I need your driver fix, I only merged the fourcc code  in libva.

HeJunyan, 2019-04-22:
Still need to add ARGB10 to surface and image query, and then, application can know such kind of format exists.

dnagasun, 2019-06-10:
hi @FurongZhang, appreciate if you could provide the status of this fix? Need media driver fix to close this issue.

FurongZhang, 2019-06-11:
I am working on that now.

FurongZhang, 2019-06-18:
https://github.com/intel/media-driver/pull/651
https://github.com/intel/media-driver/pull/652
https://github.com/intel/media-driver/pull/653

FurongZhang, 2019-06-18:
I have already submitted these 3 PRs, please help to verify if RGB10 support is ready or not in driver. If not, please let me know. If yes, please close this issue.

FurongZhang, 2019-06-18:
And glimagesink will not support 10bit in the near future in GStreamer.

xhaihao, 2019-06-21:
The latest glimagesink may accept P010_10LE/BGR10A2_LE/RGB10A2_LE on little endian and do down-sampling internally if the framebuffer is 8bit.

HeJunyan, 2019-06-27:
the #651 really fix query problem.
I am sorry I made a mistake.
After check, I find the Media_Format_R10G10B10A2 is only supported in DdiMedia_DeriveImage.
But for DdiMedia_CreateImage and GET/PUT functions, we just get VA_STATUS_ERROR_UNIMPLEMENTED.
All the images related functions need to add Media_Format_R10G10B10A2 support.
The logic in  gst-vaapi will create the image first and upload/download the images to surface.
while the logic in msdk seems directly derive the image from surface and use it.
That's the difference make the msdk can work while gstreamer-vaapi can not.

FurongZhang, 2019-07-04:
We have already added 10bit RGB support in DdiMedia_CreateImage and GET/PUT functions. The code fixes have been merged. Please check and let us know if any issues, thanks a lot in advance!

HeJunyan, 2019-07-08:
Thanks, and verified.

closed 2019-07-08 by HeJunyan