intel/media-driver #540: [H264][DEC] Incorrected resolutions return

aulitin, 2019-02-28:
Found problem with h264 decode on BDW Ubuntu 18.04.. Function vaQueryConfigEntrypoints don't work correctly. This function returns 8192x8192 resolution for VAProfileH264High profile and VAEntrypointVLD entrypoint. But [it](https://github.com/intel/media-driver/blob/master/docs/media_features.md) says they the output resolution is 4K.
Commit 0e46d8574ca07d4580c20dadefd57a005213855a introduced regression. 
Based on the code this incorrect output may be found on other platforms.

XinfengZhang, 2019-03-01:
thanks report, yes, BDW AVC does not support level 6.x, @wangyan42164 , please help to check it

wangyan42164, 2019-03-03:
I will check it.

aulitin, 2019-03-21:
@wangyan42164, Can your pull request fix [this problem](https://github.com/intel/media-driver/issues/550)?
I am waiting for fix. When are you going to merge pull request?

PR #545 (2019-06-10): 8K resolution size is only for HEVC not for AVC and VP9.
Fixes #540.
commit 475f0a87 (2019-06-10): 8K resolution size is only for HEVC and VP9 not for AVC.

Fixes #540.
closed 2019-06-10 by intel-mediadev