intel/media-driver #549: vaGetConfigAttributes for VAConfigAttribEncMaxRefFrames and HAVC returns VA_STATUS_ERROR_INVALID_PARAMETER

rosetta-jpn, 2019-03-06:
vaGetConfigAttributes for VAConfigAttribEncMaxRefFrames and HAVC profile will return
(with intel-vaapi-driver)
VA_STATUS_SUCCESS
value=(1 << 16) | (4 << 0) 
https://github.com/intel/intel-vaapi-driver/blob/72f10f16f7e5767492acde130eeeaa598d26a3a6/src/i965_drv_video.c#L1109
(with media-driver)
VA_STATUS_ERROR_INVALID_PARAMETER
value=N/A
https://github.com/intel/media-driver/blob/eb0ffe9fd0ea33dfc28f23a6fdc8217a22aa46ee/media_driver/linux/gen8/ddi/media_libva_caps_g8.cpp#L43
https://github.com/intel/media-driver/blob/eb0ffe9fd0ea33dfc28f23a6fdc8217a22aa46ee/media_driver/linux/gen9/ddi/media_libva_caps_g9.cpp#L42

media-driver should return VA_STATUS_SUCCESS with a meaningful value as well as intel-vaapi-driver.

uartie, 2019-03-06:
Oddly when running the GetConfigAttributes tests in libva-utils (https://github.com/intel/libva-utils/blob/e8045a4fd83193d4c29efdb54253951d25e00a84/test/test_va_api_config_attribs.cpp#L101 ) they don't seem to complain.

uartie, 2019-03-06:
Which VAProfile and VAEntrypoint are you using?

sreerenjb, 2019-03-06:
I believe it is taken care of by this code: https://github.com/intel/media-driver/blob/eb0ffe9fd0ea33dfc28f23a6fdc8217a22aa46ee/media_driver/linux/common/ddi/media_libva_caps.cpp#L600

@rosetta-jpn  Do you have any test case which is failing?

rosetta-jpn, 2019-03-16:
Sorry for delay response. I don't have any test.
The value is filled in media_libva_caps.cpp then?
The above g8 and g9 code is unreachable?

Xiaogangli-intel, 2019-09-04:
@rosetta-jpn 
https://github.com/intel/intel-vaapi-driver/blob/72f10f16f7e5767492acde130eeeaa598d26a3a6/src/i965_drv_video.c#L1109
is for H264.
Gen8 HEVC should be at https://github.com/intel/intel-vaapi-driver/blob/72f10f16f7e5767492acde130eeeaa598d26a3a6/src/i965_drv_video.c#L1101
Gen9 at https://github.com/intel/intel-vaapi-driver/blob/72f10f16f7e5767492acde130eeeaa598d26a3a6/src/i965_drv_video.c#L1113

And we are align. The above g8/g9 code could be reached, you should notice the condition
https://github.com/intel/media-driver/blob/eb0ffe9fd0ea33dfc28f23a6fdc8217a22aa46ee/media_driver/linux/gen9/ddi/media_libva_caps_g9.cpp#L42-L52
only when entrypoint is VAEntrypointEncSliceLP and not HEVC will return VA_STATUS_ERROR_INVALID_PARAMETER.

So this is not a issue, I will close it, if you still have questions, please let me know and reopen it.

closed 2019-09-04 by Xiaogangli-intel