intel/media-driver #550: [HEVC][ENC] Incorrected resolutions return

aulitin, 2019-03-06:
Found problem with hevc encode on SKL Ubuntu 18.04.. Function vaQueryConfigEntrypoints don't work correctly. This function returns 8192x8192 resolution for VAProfileHEVCMain profile and VAEntrypointEncSlice entrypoint. But it says they the output resolution is 4K.
Version Media-drver: 224141564fe20c19e279ec004f89fa92534f90a0

semelianova, 2019-04-24:
The same problem appears on CFL and KBL. 8k is not supported on platform less ICL.

dmitryermilov, 2019-04-24:
@aulitin , @semelianova , would you like to prepare a pool request with a fix to the driver repo?

aulitin, 2019-04-24:
@dmitryermilov, I am unable to do it soon.

Xiaogangli-intel, 2019-09-04:
@aulitin [the table](https://github.com/intel/media-driver/blob/master/docs/media_features.md) was updated, the output resolution should be 8K, could you confirm it?

semelianova, 2019-09-04:
@Xiaogangli-intel Is the table correct? When try to encode or decode 8K resolution stream I get VA_STATUS_ERROR_RESOLUTION_NOT_SUPPORTED from driver caps

semelianova, 2019-09-04:
I think need to add HEVC profile with 8k resolution in  [CheckEncodeResolution ](https://github.com/intel/media-driver/blob/b6e7cc5cdd5e1a306c47e1bf48952e381712b414/media_driver/linux/common/ddi/media_libva_caps.cpp#L1944) function. Now we fall in default case and compare with 4K max.

semelianova, 2019-10-15:
Hi @Xiaogangli-intel, Are there any updates? Now HEVCe doesn't work with 8k. Should HEVCe support 8k resolution on gen9 platforms? Does driver really support it?

PR #838 (2020-02-19): [Encode][HEVC] Fix max width and height caps for gen9 platforms
According to: https://github.com/intel/media-driver/blob/master/docs/media_features.md
Also in [MediaLibvaCaps::CheckEncodeResolution](https://github.com/intel/media-driver/blob/2711a3a1c3339b1a5f5d5e1c76fccd24f1998d81/media_driver/linux/common/ddi/media_libva_caps.cpp#L1954) function we check the max size as 4k.

Fixes #550
commit fd169910 (2020-02-19): [Encode][HEVC] Fix max width and height caps for gen9 platforms

According to: https://github.com/intel/media-driver/blob/master/docs/media_features.md

Fixes #550
closed 2020-02-19 by intel-mediadev