intel/media-driver #556: Need to add vpp deinterlace  weave and MotionCompensated support

zhuqingliang, 2019-03-13:
reproduce step:
$ ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v debug -f rawvideo -pix_fmt yuv420p -s:v 720x480 -top 1 -i MotionTest_i_tff_720x480_10f_I420.yuv -vf 'format=nv12,hwupload,deinterlace_vaapi=mode=weave:rate=field,hwdownload,format=nv12' -pix_fmt yuv420p -vframes 20 -y dump.yuv

message:
[Parsed_deinterlace_vaapi_2 @ 0xd69e80] Deinterlacing mode 2 (Weave) is not supported.
[Parsed_deinterlace_vaapi_2 @ 0xd69e80] Failed to configure output pad on Parsed_deinterlace_vaapi_2
Error reinitializing filters!
Failed to inject frame into filter network: Invalid argument
Error while processing the decoded data for stream #0:0

$ ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v debug -f rawvideo -pix_fmt yuv420p -s:v 720x480 -top 1 -i /root/media/testing/assets/yuv/MotionTest_i_tff_720x480_10f_I420.yuv -vf 'format=nv12,hwupload,deinterlace_vaapi=mode=motion_compensated:rate=field,hwdownload,format=nv12' -pix_fmt yuv420p -vframes 20 -y dump.yuv

message:
[Parsed_deinterlace_vaapi_2 @ 0x1bfbf00] Deinterlacing mode 4 (MotionCompensated) is not supported.
[Parsed_deinterlace_vaapi_2 @ 0x1bfbf00] Failed to configure output pad on Parsed_deinterlace_vaapi_2
Error reinitializing filters!
Failed to inject frame into filter network: Invalid argument
Error while processing the decoded data for stream #0:0

Jister, 2019-06-28:
Actually, driver internally use motion compensated when app set VAProcDeinterlacingMotionAdaptive, it's a bug, so I will correct the capability from VAProcDeinterlacingMotionAdaptive to VAProcDeinterlacingMotionCompensated.
Since MCDI has better quality than MADI, so we just enable MCDI, and won't enable MADI currently. And we also won't enable VAProcDeinterlacingWeave. 
Do you have a strong requirement to enable MADI/Weave? If so, could you explain in what situations will you use these modes? Thanks.

XinfengZhang, 2020-07-06:
already was fixed by 60cdeaaee875e7b656c12688fb204a54c5b7bb32

closed 2020-07-06 by XinfengZhang