intel/media-driver #558: [gstreamer-vaapi][hevc 10bit decode][regression]all cases dumped files have garbage and not bit match with reference md5 on APL

wangzj0601, 2019-03-14:
System Environment
=======

Platform: APL
Arch: x86_64
Kernel: 4.18.0+
libva commit 0dc70068e49f9538fbd03f2f0aa1e56a91e986e7

gmmlib commit ac68e2cebf9774b17bc8a4057d30959117222127
media_driver commit 7a35908c575306d3764dfb98e8e6b83d185ffea1
gstreamer master branch e322250fb1d1c31692d1e97e3de19ead6b016d7f
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer.git

gst-plugins-base commit b0faf032d20456b3006bf33c79db30ceb96d6ea8

repo: https://gitlab.freedesktop.org/gstreamer/gst-plugins-base.git
gst-vaapi master branch a90bf00906ef6663ac0d3701c19043c72835f824
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi.git

Regression?
=======

Yes

Bug detailed description
=======

all cases have garbage and not bit match with reference md5 on APL

this issue can not duplicated on KBL

Reproduce Steps
==============

1.build enc as above lists

2. gst-launch-1.0 -vf filesrc location=/media_stream/hevc/10bit/input_hevc_10bit.bin ! h265parse ! vaapih265dec ! videoconvert ! video/x-raw,format=P010_10LE ! checksumsink2 file-checksum=false qos=false frame-checksum=false plane-checksum=false dump-output=true dump-location=./test.yuv

3. the dumped file has garbage

Expected Result
=============
the dumped file has no garbage, and bit match with ref md5

wangzj0601, 2019-03-14:
regression commit:

d7e8c7c sw swizzle surface in vaGetImage

this issue can not be duplicated with gst-msdk on APL.

XinfengZhang, 2020-05-26:
@wangzj0601 , there are some changes in vaGetImage, could you help to check it again? now , we are using VPP to handle it

XinfengZhang, 2020-07-02:
could be closed, current flow in vaGetImage is: create linear surface, call VPP to do de-swizzle, then copy to system memory.

closed 2020-07-02 by XinfengZhang