intel/media-driver #560: media driver doesn't support WHL with device id 0x3EA0

TianmiChen, 2019-03-14:
We see that media driver (https://github.com/intel/media-driver/blob/master/README.md ) should support WHL (Whiskey Lake) platform. But we cannot have video playback work on WHL and got below error messages when loading the driver. We checked on Bspec that device id 3ea0 is WHL. Pease confirm whether the driver supports WHL with device iD 3ea0 or not.

///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
2-31 19:41:25.920 13530 13530 I libva   : VA-API version 1.2.0
12-31 19:41:25.920 13530 13530 I libva   : va_getDriverName() returns 0
12-31 19:41:25.920 13530 13530 I libva   : Trying to open /vendor/lib64//i965_drv_video.so
12-31 19:41:25.923 13530 13530 I libva   : Found init function __vaDriverInit_1_2
12-31 19:41:25.924 13530 13530 E MOS     : [MOS]:  CRITICAL - HWInfo_GetGfxInfo:108: Failed to get the device info for Device id: 3ea0
12-31 19:41:25.924 13530 13530 E MOS     :
12-31 19:41:25.924 13530 13530 E MOS     : [LIBVA]:CRITICAL - DdiMedia__Initialize:1303: Fatal error - unsuccesfull Sku/Wa/GtSystemInfo initialization
12-31 19:41:25.924 13530 13530 E libva   : /vendor/lib64//i965_drv_video.so init failed
12-31 19:41:25.924 13530 13530 I libva   : va_openDriver() returns 1
//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

 
Thanks,

dvrogozh, 2019-03-14:
You are loading another driver: i965_drv_video.so, which is this one: https://github.com/intel/intel-vaapi-driver. As far as I know it has limited WHL support.

To load iHD, make sure to have:
`export LIBVA_DRIVER_NAME=iHD`

TianmiChen, 2019-03-15:
Thanks for pointing out! After switched to VPG driver, this issue is gone.

closed 2019-03-25 by XinfengZhang