intel/media-driver #561: README file wrong or not updated correctly?

makedir, 2019-03-15:
https://github.com/intel/media-driver/blob/master/README.md#hw-media-features

Why does it claim, Coffee Lake doesnt support HEVC 8 and 10bit hw encode?

[image]

Entropy512, 2019-06-11:
If I'm reading the documentation correctly, those platforms require nonfree shaders in order to encode those particular codecs.  You've linked the table for the fully opensource version of the driver.  Scroll up for the table of codecs supported with nonfree shaders.

Sherry-Lin, 2019-06-12:
Yes, we have two build options. Non-free kernel is default option which could support full features(https://github.com/intel/media-driver#features-in-default-driver-build), and free kernel is mapping to full open source build(https://github.com/intel/media-driver#features-in-open-source-build).

XinfengZhang, 2020-06-07:
already updated it https://github.com/intel/media-driver#components-and-features

closed 2020-06-07 by XinfengZhang