intel/media-driver #563: [gstreamer-vaapi][avc cqp encode]the cmdline will hang up when run it several times on APL

wangzj0601, 2019-03-15:
System Environment
=======
Platform: APL
Arch: x86_64
Kernel: 4.18.0+
libva commit 0dc70068e49f9538fbd03f2f0aa1e56a91e986e7
gmmlib commit ac68e2cebf9774b17bc8a4057d30959117222127
media_driver commit 7a35908c575306d3764dfb98e8e6b83d185ffea1
gstreamer master branch e322250fb1d1c31692d1e97e3de19ead6b016d7f
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer.git
gst-plugins-base commit b0faf032d20456b3006bf33c79db30ceb96d6ea8
repo: https://gitlab.freedesktop.org/gstreamer/gst-plugins-base.git
gst-vaapi master branch a90bf00906ef6663ac0d3701c19043c72835f824
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi.git

Bug detailed description
=======
the cmdline will hang up when run it several times on APL
this issue can not be duplicated on KBL/SKL

failed cases:
test_cqp(bframes=2,case=bluesky,gop=30,profile=high,qp=10,quality=1,slices=1) TIMEOUT FAIL
test_cqp(bframes=2,case=bluesky,gop=30,profile=high,qp=10,quality=1,slices=3) TIMEOUT FAIL
test_cqp(bframes=2,case=bluesky,gop=30,profile=high,qp=10,quality=4,slices=1) TIMEOUT FAIL
test_cqp(bframes=2,case=bluesky,gop=30,profile=high,qp=10,quality=4,slices=3) TIMEOUT FAIL
test_cqp(bframes=2,case=bluesky,gop=30,profile=main,qp=10,quality=1,slices=1) TIMEOUT FAIL
test_cqp(bframes=2,case=bluesky,gop=30,profile=main,qp=10,quality=1,slices=3) TIMEOUT FAIL
test_cqp(bframes=2,case=bluesky,gop=30,profile=main,qp=10,quality=4,slices=1) TIMEOUT FAIL
test_cqp(bframes=2,case=bluesky,gop=30,profile=main,qp=10,quality=4,slices=3) TIMEOUT FAIL
test_cqp(bframes=2,case=mobile,gop=30,profile=high,qp=10,quality=4,slices=1) TIMEOUT FAIL
test_cqp(bframes=2,case=mobile,gop=30,profile=high,qp=10,quality=4,slices=3) TIMEOUT FAIL
test_cqp(bframes=2,case=mobile,gop=30,profile=main,qp=10,quality=4,slices=1) TIMEOUT FAIL
test_cqp(bframes=2,case=mobile,gop=30,profile=main,qp=10,quality=4,slices=3) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=high,qp=10,quality=4,slices=1) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=high,qp=10,quality=4,slices=3) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=high,qp=10,quality=7,slices=1) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=high,qp=10,quality=7,slices=3) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=high,qp=16,quality=4,slices=1) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=high,qp=16,quality=4,slices=3) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=high,qp=16,quality=7,slices=1) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=high,qp=16,quality=7,slices=3) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=main,qp=10,quality=4,slices=1) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=main,qp=10,quality=4,slices=3) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=main,qp=10,quality=7,slices=1) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=main,qp=10,quality=7,slices=3) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=main,qp=16,quality=4,slices=1) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=main,qp=16,quality=4,slices=3) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=main,qp=16,quality=7,slices=1) TIMEOUT FAIL
test_cqp(bframes=2,case=stockholm,gop=30,profile=main,qp=16,quality=7,slices=3) TIMEOUT FAIL

Reproduce Steps
==============
1.build enc as above lists
2. gst-launch-1.0 -vf filesrc location=/media_stream/yuv/MOBILE_352x288_30_orig_01_352x288_I420.yuv num-buffers=300 ! rawvideoparse format=i420 width=352 height=288 ! videoconvert ! video/x-raw,format=NV12 ! vaapih264enc rate-control=cqp keyframe-period=30 init-qp=10 quality-level=4 num-slices=3 max-bframes=2 ! video/x-h264,profile=main ! h264parse ! filesink location=./test.h264
3. run the above cmdline again and again

Expected Result
=============
the cmdline always can run successfully quickly and no hang

Actual Result
===========
the cmdline will hang up when run it several times on APL

Xiaogangli-intel, 2020-07-09:
This issue has been fixed in the latest driver. Please help to confirm it.

xinyudox, 2020-07-14:
I tried a few cases with last media driver on APL, and here's the result.
bframes=2,case=bluesky,gop=30,profile=high,qp=10,quality=1,slices=1    timeout  Total duration: 00:05:01
bframes=2,case=bluesky,gop=30,profile=high,qp=10,quality=1,slices=3    timeout  Total duration: 00:05:01
bframes=2,case=bluesky,gop=30,profile=high,qp=10,quality=4,slices=1    timeout  Total duration: 00:05:01

Xiaogangli-intel, 2020-07-15:
@xinyudox could you list the versions of the tool list and the ubuntu version? Also, could you provide dmesg?
Thank you.

xinyudox, 2020-07-15:
APL | OS: ubuntu 18.04 | kernel 5.7.0-rc5-daily-nocodec-20200519
[dmesg.log](https://github.com/intel/media-driver/files/4923911/dmesg.log)

libva | afca400d5629409ae7d51c4b1109a1b73ab2844a
Mediasdk | 04626ad4ca5d720957cbe59fd7b468d978e36150
gmmlib | 40c7d14fd16a4282410a2f6b0cd59251c8e41d5c
iHD Driver | 388ea758af1f050b66c9150f7981192cea0ae540
libva-utils | 287f3a581f46f2e38aa7dc8b371033e91110da6a
gst-plugins-bad | 495ed45d056cc4ae88c95b5587732d87382e93a1
gst-plugins-base | 93cb325fa133d494369761b029b0583a2ff66605
gstreamer-vaapi | b1832223ffb72360e01fe30cdf0b88423b0455b2
gst-plugins-good | 5a7275220bd18cb60b8f91ccadf919554de68383
gstreamer | cecb2912633164a66a942d5413404812ad8fc0a0
gst-checksumsink | 766a46d628a78345de6ac6fe932dd2c3a80d59b4
vaapi-fits-full | 14fdb2a083fbc1e45ff2fc6a428d494110ee3277
vaapi-fits | 950b99722a387cba1ce30c72f7c514d146ff0083

Xiaogangli-intel, 2020-07-16:
From dmesg, we didn't find hang. Synced with @FocusLuo, as this issue cannot be reproduced on my side based on the latest tool list, I will close this now. If it still exists on your side, please issue a new one.

closed 2020-07-16 by Xiaogangli-intel