intel/media-driver #564: [gstreamer-vaapi][vpp][denoise/scale/contrast] the output has garbage if the output format is NV12 on APL

wangzj0601, 2019-03-15:
System Environment
=======
Platform: APL
Arch: x86_64
Kernel: 4.18.0+
libva commit 0dc70068e49f9538fbd03f2f0aa1e56a91e986e7
gmmlib commit ac68e2cebf9774b17bc8a4057d30959117222127
media_driver commit 7a35908c575306d3764dfb98e8e6b83d185ffea1
gstreamer master branch e322250fb1d1c31692d1e97e3de19ead6b016d7f
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer.git
gst-plugins-base commit b0faf032d20456b3006bf33c79db30ceb96d6ea8
repo: https://gitlab.freedesktop.org/gstreamer/gst-plugins-base.git
gst-vaapi master branch a90bf00906ef6663ac0d3701c19043c72835f824
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi.git

Bug detailed description
=======
when do vpp denoise/scale/contrast, the output has garbage if the output format is NV12 on APL
this issue can not duplicated when do vpp denoise/scale/contrast, the output has garbage if the output format is I420 on APL
this issue can not be duplicated on KBL/SKL

Reproduce Steps
==============
1.build enc as above lists
2. scale + NV12 output format
gst-launch-1.0 -vf filesrc location=/media_stream/yuv/1080p_blue_sky_1920x1080_NV12.yuv num-buffers=10 ! rawvideoparse format=nv12 width=1920 height=1080 ! vaapipostproc format=nv12 width=640 height=480 ! filesink location=./bluesky_scaled_640x480_NV12.yuv

3. denoise + NV12 output format
gst-launch-1.0 -vf filesrc location=/media_stream/yuv/1080p_blue_sky_1920x1080_NV12.yuv num-buffers=10 ! rawvideoparse format=nv12 width=1920 height=1080 ! vaapipostproc format=nv12 denoise=0.5 ! filesink location=./1080p_blue_sky_denoise_50_NV12_1920x1080.yuv

4.contrast + NV12 output format
gst-launch-1.0 -vf filesrc location=/media_stream/yuv/1080p_blue_sky_1920x1080_NV12.yuv num-buffers=10 ! rawvideoparse format=nv12 width=1920 height=1080 ! vaapipostproc format=nv12 contrast=2.0 ! filesink location=./bluesky_contrast_100_NV12_1920x1080.yuv

Expected Result
=============
when do vpp denoise/scale/contrast, the output has no garbage even though the output format is NV12 on APL

FurongZhang, 2019-06-18:
Can we reproduce this issue on KBL? I don't have APL, I only have one KBL. If it can be reproduced on KBL, I can debug that.

KamistixX, 2019-06-18:
> 
> 
> Can we reproduce this issue on KBL? I don't have APL, I only have one KBL. If it can be reproduced on KBL, I can debug that.

It does not seem to be reproducible on Kaby Lake unfortunately, only Apollo Lake. It seems to be the same issue as reported [here](https://github.com/intel/intel-vaapi-driver/issues/398)

closed 2022-11-18 by FurongZhang