intel/media-driver #576: [ffmpeg-vaapi][hevc Y210 decode][ICL] GPU Hung

fulinjie, 2019-03-25:
Apply the patch:
https://github.com/intel-media-ci/ffmpeg/pull/27/commits/5dfd317058e3240eeeb83729c38c4320fb6c3682

CMDLine:
ffmpeg -v debug -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -y -i ./Main_422_10_A_RExt_Sony_2.bin -f rawvideo y210.yuv

Error message:
[AVHWFramesContext @ 0x7f49742367b0] Map surface 0x18.
[hevc @ 0x55efb50333d0] nal_unit_type: 2(TSA_N), nuh_layer_id: 0, temporal_id: 1
[hevc @ 0x55efb50333d0] nal_unit_type: 40(SEI_SUFFIX), nuh_layer_id: 0, temporal_id: 1
[hevc @ 0x55efb50333d0] Output frame with POC 4.
[hevc @ 0x55efb50333d0] Param buffer (type 0, 604 bytes) is 0x2.
[hevc @ 0x55efb50333d0] Decoding SEI
[hevc @ 0x55efb50333d0] Slice 0 param buffer (264 bytes) is 0x1.
[hevc @ 0x55efb50333d0] Slice 0 data buffer (50732 bytes) is 0.
[hevc @ 0x55efb50333d0] Decode to surface 0x11.
[AVHWFramesContext @ 0x7f49742367b0] **Failed to sync surface 0x18: 34 (HW busy now).**
[hevc @ 0x55efb4fcf630] Failed to transfer data to output frame: -5.
Error while processing the decoded data for stream #0:0
[AVIOContext @ 0x55efb4fcfb80] Statistics: 0 seeks, 0 writeouts
[AVIOContext @ 0x55efb4fd5c60] Statistics: 1943569 bytes read, 0 seeks

dmesg:
[318970.575931] audit: type=1006 audit(1553475965.488:63): pid=5705 uid=0 old-auid=4294967295 auid=1000 tty=(none) old-ses=4294967295 ses=8 res=1
[318970.599291] audit: type=1130 audit(1553475965.512:64): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=user@1000 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[321250.461417] audit: type=1006 audit(1553478245.375:65): pid=9776 uid=0 old-auid=4294967295 auid=1000 tty=(none) old-ses=4294967295 ses=9 res=1
[324358.817772] i915 0000:00:02.0: Resetting vcs0 for hang on vcs0
[324857.044534] audit: type=1006 audit(1553481851.955:66): pid=28686 uid=0 old-auid=4294967295 auid=1000 tty=(none) old-ses=4294967295 ses=10 res=1
[324911.777703] i915 0000:00:02.0: Resetting vcs0 for hang on vcs0
[326285.581854] audit: type=1006 audit(1553483280.491:67): pid=29065 uid=0 old-auid=4294967295 auid=1000 tty=(none) old-ses=4294967295 ses=11 res=1
[329287.883333] i915 0000:00:02.0: **Resetting vcs0 for hang on vcs**

Trace the position of GPU Hang:
https://github.com/intel/media-driver/blob/91e6176a9efa0a529194b28180469d60a9cb3ea7/media_driver/linux/common/os/mos_os_specific.c#L5979

in Mos_Specific_IsGPUHung(),
dwActiveBatch != pOsInterface->dwGPUActiveBatch

wangyan-intel, 2019-04-11:
@fulinjie could you share your input stream for debugging?

fulinjie, 2019-04-26:
ping?
Any update?

wangyan-intel, 2019-04-26:
@fulinjie I have sent mail to Fei about #601, #602, #603. I think this issue may be similar with them. You can talked with Fei at first. Thanks.

fulinjie, 2019-04-26:
@wangyan-intel  This situation is kind of different, the hang issue occurs during map and sync surface, which didn't call parsePicParams yet.

wangyan-intel, 2019-04-26:
I will check it. Thanks.

wangyan42164, 2019-04-28:
@fulinjie I checked ffmpeg code. It also hasn't VAPictureParameterBufferHEVCRext and only has VAPictureParameterBufferHEVC. And after debugging, I found GPU hang occurs after the first call of DdiDecodeHEVCG11::ParsePicParams(). (It should be the 1st frame. GPU hang should occurs after finish updating all parameters include picture into HW.)
And Fei has confirmed #602 and #603 will be fixed after fill VAPictureParameterBufferHEVCRext rightly. Could you please double check it?
I will also continue to debug it.
Thanks.

fulinjie, 2019-05-08:
Thanks a lot for the input @wangyan-intel , ffmpeg vaapi decode works for 422/422_10/444/444_10 streams now with patch set:
https://github.com/intel-media-ci/ffmpeg/pull/41

closed 2019-05-08 by fulinjie