intel/media-driver #578: [ffmpeg-vaapi][VP8 encode] floating point exception

eero-t, 2019-03-29:
Setup:
* Tested to happen both on ICL and KBL GT2
* Git version of drm-tip kernel and all media components

Use-case:
* ffmpeg -loglevel verbose -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -threads 1 -i 1280x720p_29.97_10mb_h264_cabac.264  -c:v vp8_vaapi  -y test.webm

Result is floating point exception:
```
...
Input #0, h264, from 'input/1280x720p_29.97_10mb_h264_cabac.264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), 1 reference frame, yuv420p(tv, bt709, progressive, left), 1280x720 [SAR 1:1 DAR 16:9], 29.97 fps, 29.97 tbr, 1200k tbn, 59.94 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> vp8 (vp8_vaapi))
Press [q] to stop, [?] for help
[h264 @ 0x5618e5b2ef40] Reinit context to 1280x720, pix_fmt: vaapi_vld
[graph 0 input from stream 0:0 @ 0x5618e6303b80] w:1280 h:720 pixfmt:vaapi_vld tb:1/1200000 fr:30000/1001 sar:1/1 sws_param:flags=2
[vp8_vaapi @ 0x5618e5b0f480] Input surface format is nv12.
[vp8_vaapi @ 0x5618e5b0f480] Using VAAPI profile VAProfileVP8Version0_3 (14).
[vp8_vaapi @ 0x5618e5b0f480] Using VAAPI entrypoint VAEntrypointEncSlice (6).
[vp8_vaapi @ 0x5618e5b0f480] Using VAAPI render target format YUV420 (0x1).
[vp8_vaapi @ 0x5618e5b0f480] No quality level set; using default (40).
[vp8_vaapi @ 0x5618e5b0f480] RC mode: CQP.
[vp8_vaapi @ 0x5618e5b0f480] RC quality: 40.
[vp8_vaapi @ 0x5618e5b0f480] RC framerate: 30000/1001 (29.97 fps).
[vp8_vaapi @ 0x5618e5b0f480] Using intra and P-frames (supported references: 3 / 0).
[vp8_vaapi @ 0x5618e5b0f480] Driver does not support any packed headers (none wanted).
Output #0, webm, to 'test.webm':
  Metadata:
    encoder         : Lavf58.26.101
    Stream #0:0: Video: vp8 (vp8_vaapi), 1 reference frame, vaapi_vld(left), 1280x720 [SAR 1:1 DAR 16:9], q=-1--1, 29.97 fps, 1k tbn, 29.97 tbc
    Metadata:
      encoder         : Lavc58.47.106 vp8_vaapi
Floating point exception
```
Same happens also with #409 use-case.

wangyan42164, 2019-05-13:
@eero-t Could you please share this bit stream for reproducing? Thanks

wangyan-intel, 2019-05-14:
@eero-t I submitted the PR#622 to fix it. It works fine on my platform. Could you please try it? Thanks.

eero-t, 2019-05-14:
Yes, it gets rid of the exception.  However, fix to #409 is also needed, otherwise output file is empty.

(Tested on KBL GT2.)

zhoub, 2019-05-20:
Interesting, I have this issue with libva+vaapi driver at i7 8700k.
```
vainfo: VA-API version: 1.4 (libva 2.4.0)
vainfo: Driver version: Intel iHD driver - 18.4.git_1d28de_2019-02-11
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
```

PR #622 (2019-05-21): Fix VP8 CQP encoding.
1. Initialize BRC variales for avoiding wrong DSH allocation.
2. Set the right CQP enum number (RATECONTROL_CQP = 3).
3. Avoid division by zero in CQP mode.
Fixes #578.
commit 80d8db42 (2019-05-21): Fix VP8 CQP encoding.

1. Initialize BRC variales for avoiding wrong DSH allocation.
2. Set the right CQP enum number (RATECONTROL_CQP = 3).
3. Avoid division by zero in CQP mode.
Fixes #578.
closed 2019-05-21 by intel-mediadev