intel/media-driver #579: [ICL][ffmpeg-vaapi] VP9 encode doesn't work

eero-t, 2019-03-29:
Setup:
* drm-tip kernel git version
* latest git versions of all media components (from libdrm to ffmepg)

Use-case is VP9 encoding: 
* ffmpeg -loglevel verbose -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i 1280x720p_29.97_10mb_h264_cabac.264 -c:v vp9_vaapi -low_power 1 -qscale:v 20 -y test.webm

It fails:
```
...
[h264 @ 0x55640aae4f00] Reinit context to 1280x720, pix_fmt: vaapi_vld
[graph 0 input from stream 0:0 @ 0x55640ab13500] w:1280 h:720 pixfmt:vaapi_vld tb:1/1200000 fr:30000/1001 sar:1/1 sws_param:flags=2
[vp9_vaapi @ 0x55640aae3f80] Input surface format is nv12.
[vp9_vaapi @ 0x55640aae3f80] Using VAAPI profile VAProfileVP9Profile0 (19).
[vp9_vaapi @ 0x55640aae3f80] No usable encoding entrypoint found for profile VAProfileVP9Profile0 (19).
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
```

This happens regardless of:
* Input resolution and whether it's 10-bit or 8-bit
* Whether "-low_power 1 -qscale:v XX" options are used or not

According to media-driver README.md, VP9 is supported on ICL with HW encoding.  There are few issues mentioned for for ICL, but I have latest drm-tip kernel and this is VP9, not AVC

Xiaogangli-intel, 2019-04-23:
Hi Eero,
          Currently, GuC and HuC firmware is not ready in drm-tip, so  HEVC/VP9 low power cases will not work! GuC/HuC up-streaming is WIP, will let you know once it ready.

eero-t, 2019-05-14:
> Currently, GuC and HuC firmware of ICL is not ready in drm-tip, so HEVC/VP9 low power cases will not work!

What about non-low-power (shader based) VP9 HW encoding?

Xiaogangli-intel, 2019-09-04:
@eero-t GuC and HuC firmware of ICL is ready in drm-tip, please confirm it. Let me know if you still have questions.

eero-t, 2019-09-05:
I don't have anymore ICL on which to test.  May take weeks until there's a replacement.

eero-t, 2019-10-08:
Media driver seems to implement only low-power encoding for VP9.  One doesn't need to specify "-low_power 1" option to FFmpeg for it to work, but one needs to have HuC for things to work:
`sudo echo "options i915 enable_guc=2" > /etc/modprobe.d/i915.conf; sudo reboot`

After that, the command line given in first comment fails to an invalid memory read:
```
...
[AVBSFContext @ 0x160fc640] Too many invisible frames
[AVBSFContext @ 0x160fc640] Failed to receive packet from filter vp9_superframe for stream 0
==2173== Invalid read of size 8
==2173==    at 0xE6EB3C: ??? (in /opt/install/bin/ffmpeg)
==2173==  Address 0x13d7dc10 is 0 bytes after a block of size 12,288 alloc'd
==2173==    at 0x4C31B25: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==2173==    by 0xDA574A2: MOS_AllocAndZeroMemory (mos_utilities.c:4093)
==2173==    by 0xDC55A9F: DdiEncodeVp9::ContextInitialize(CodechalSetting*) (media_ddi_encode_vp9.cpp:362)
==2173==    by 0xDC3C8FE: DdiEncode_CreateContext(VADriverContext*, unsigned int, int, int, int, unsigned int*, int, unsigned int*) (media_libva_encoder.cpp:365)
==2173==    by 0x679E24D: vaCreateContext (va.c:1194)
==2173==    by 0x27FFD2: ??? (in /opt/install/bin/ffmpeg)
==2173== 
[AVBSFContext @ 0x160fc640] Too many invisible frames
[AVBSFContext @ 0x160fc640] Failed to receive packet from filter vp9_superframe for stream 0
==2173== Invalid read of size 4
==2173==    at 0xE6EB49: ??? (in /opt/install/bin/ffmpeg)
==2173==  Address 0x1000100000008 is not stack'd, malloc'd or (recently) free'd
==2173== 
==2173== Process terminating with default action of signal 11 (SIGSEGV)
==2173==  General Protection Fault
```

If one replaces vaapi_vp9 with vaapi_vp8, everything works fine.

mattburgess, 2020-04-28:
I've just purchased an ICL-based laptop with the hopes of being able to encode VP9 with hardware acceleration, but appear to have run into this bug!

This is on a freshly installed Fedora 32 OS with the following components:

```
intel-gmmlib.x86_64                   20.1.1-1.fc32
intel-media-driver.x86_64             20.1.1-1.fc32
intel-mediasdk.x86_64                 20.1.0-1.fc32
kernel.x86_64                         5.6.6-300.fc32 
libva.x86_64                          2.7.1-1.fc32
libva-intel-driver.x86_64             2.4.0-3.fc32
libva-intel-hybrid-driver.x86_64      1.0.2-18.fc32
```

`vainfo` shows that encoding support isn't available:

```
$ vainfo | grep VP9
libva info: VA-API version 1.7.0
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_7
libva info: va_openDriver() returns 0
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile1            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileVP9Profile3            :	VAEntrypointVLD
```
And this command, ultimately is the reproducer that fails:

```
ffmpeg -vaapi_device /dev/dri/renderD128 -i input.mpeg -vf 'format=nv12,hwupload' -c:v vp9_vaapi -b:v 5M output.webm
ffmpeg version 4.2.2 Copyright (c) 2000-2019 the FFmpeg developers
...
[vp9_vaapi @ 0x55bbd0d5b900] No usable encoding entrypoint found for profile VAProfileVP9Profile0 (19).
```
Is this a known upstream bug, or is this a Fedora packaging issue? If it's the latter, any hints as to which component is missing support would be much appreciated please.

mattburgess, 2020-04-29:
Apologies; this does appear to work but requires the GUC/HUC firmware to be loaded as @eero-t mentions above.

eero-t, 2020-05-04:
Can driver detect whether HuC firmware is loaded or not?  If it can, I think it should give a specific warning/error about that, telling how to fix that.

Xiaogangli-intel, 2020-06-24:
@eero-t @mattburgess 
If HuC was not loaded successfully, you will not get VAEntrypointEncSliceLP for VP9 profiles. If HuC loaded successfully, the information in vainfo should like this:
[image]

Please check your GuC and HuC load status using the following command:
cat /sys/kernel/debug/dri/0/i915_guc_load_status
cat /sys/kernel/debug/dri/0/i915_huc_load_status

The output should like this:
# cat /sys/kernel/debug/dri/0/i915_guc_load_status
GuC firmware: i915/icl_guc_33.0.0.bin
        status: RUNNING
        version: wanted 33.0, found 33.0
        uCode: 384896 bytes
        RSA: 256 bytes

GuC status 0x800330ec:
        Bootrom status = 0x76
        uKernel status = 0x30
        MIA Core status = 0x3

Scratch registers:
         0:     0x0
         1:     0x0
         2:     0x1dafd3
         3:     0x0
         4:     0x4000
         5:     0x40
         6:     0x3da
         7:     0x0
         8:     0x0
         9:     0x0
        10:     0x0
        11:     0x0
        12:     0x0
        13:     0x0
        14:     0x0
        15:     0x0

k# cat /sys/kernel/debug/dri/0/i915_huc_load_status
HuC firmware: i915/icl_huc_9.0.0.bin
        status: RUNNING
        version: wanted 9.0, found 9.0
        uCode: 498496 bytes
        RSA: 256 bytes

HuC status 0x00000000:

Xiaogangli-intel, 2020-06-24:
> Media driver seems to implement only low-power encoding for VP9. One doesn't need to specify "-low_power 1" option to FFmpeg for it to work, but one needs to have HuC for things to work:
> `sudo echo [...]

@eero-t, I cannot reproduce this issue using the latest media-driver, could you try the latest driver? If it still doesn't work, could you provide GuC and HuC status?

Xiaogangli-intel, 2020-07-02:
@eero-t do you have time for a try?

Xiaogangli-intel, 2020-07-09:
This issue has been fixed in the latest driver, so close it.

closed 2020-07-09 by Xiaogangli-intel

eero-t, 2020-07-13:
Sorry for late reply.  I don't have ICL HW anymore, so unfortunately I cannot verify this.

Maybe media driver could output something like this when required functionality is missing: "HuC firmware is required for functionality <X>, check from "/sys/kernel/debug/dri/0/i915_huc_load_status" whether it's loaded, and if not, add kernel option to do that"...?

XinfengZhang, 2020-07-14:
@Xiaogangli-intel hi Xiaogang, could we add some message for it, then there will be some clue to facilitate the debug