intel/media-driver #582: ffmpeg-vaapi can't specify SFC scaling mode

lizhong1008, 2019-04-01:
Summary:  
    There is no any difference to set scaling mode as fast (VA_FILTER_SCALING_FAST) and hq (VA_FILTER_SCALING_HQ).  The output files are same, the performances are same, GPU usage are same. 

Expectation: VA_FILTER_SCALING_FAST should be faster，lower GPU usage than VA_FILTER_SCALING_HQ.

Reproduce:
1. ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i /root/lizhong/bbb_sunflower_1080p_30fps_normal.mp4 -vframes 1000 -vf 'scale_vaapi=w=1280:h=720:mode=hq' -c:v h264_vaapi -low_power 1 -q 25 output_hq.mp4 -v verbose

fps=154
GPU usage: 
./metrics_monitor
RENDER usage: 13.13,    VIDEO usage: 38.39,     VIDEO_E usage: 0.00     GT Freq: 1125.63

2. ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i /root/lizhong/bbb_sunflower_1080p_30fps_normal.mp4 -vframes 1000 -vf 'scale_vaapi=w=1280:h=720:mode=fast' -c:v h264_vaapi -low_power 1 -q 25 output_fast.mp4 -v verbose
fps=154
GPU usage is very close as hq mode

diff test_scale_lp.mp4 test_scale_hq.mp4
They are same. 

Something more: when set scaling mode to be VA_FILTER_SCALING_DEFAULT, the RENDER usage is 0, and fps=159.  So VA_FILTER_SCALING_DEFAULT is SFC?

Check code:
https://github.com/intel/media-driver/blob/master/media_driver/linux/common/vp/ddi/media_libva_vp.c#L1114 
    {
    case VA_FILTER_SCALING_FAST:
        if (pVpHalSrcSurf->SurfType == SURF_IN_PRIMARY)
        {
            pVpHalSrcSurf->ScalingMode       = VPHAL_SCALING_AVS;
            pVpHalSrcSurf->ScalingPreference = VPHAL_SCALING_PREFER_SFC_FOR_VEBOX;
        }

I believe the problem should be SFC is not used if no VEBOX pipeline.

lizhong1008, 2019-04-02:
@FurongZhang @XinfengZhang

lizhong1008, 2019-05-07:
@FurongZhang ,any update?

FurongZhang, 2019-05-07:
1.  Regarding VA_FILTER_SCALING_FAST, if no VEBOX specific features are enabled and VAAPI user set VA_FILTER_SCALING_FAST, driver will use Render Engine(AVS sampler) to do scaling which is same as VA_FILTER_SCALING_HQ.
2.  VA_FILTER_SCALING_DEFAULT is using SFC if not hitting HW limitation.
Code details, please see the below: 
-----------------------------------------------------------------------------------------------------------------
case VA_FILTER_SCALING_FAST:
        if (pVpHalSrcSurf->SurfType == SURF_IN_PRIMARY)
        {
            pVpHalSrcSurf->ScalingMode       = VPHAL_SCALING_AVS;
            pVpHalSrcSurf->ScalingPreference = VPHAL_SCALING_PREFER_SFC_FOR_VEBOX;
        }
But ..
    // if ScalingPreference == VPHAL_SCALING_PREFER_SFC_FOR_VEBOX, use SFC only when VEBOX is required
    if ((pSrcSurface->ScalingPreference == VPHAL_SCALING_PREFER_SFC_FOR_VEBOX)    &&
        (bVeboxNeeded == false))
    {
        VPHAL_RENDER_NORMALMESSAGE("DDI choose to use SFC only for VEBOX, and since VEBOX is not required, change to Composition.");
        pRenderData->OutputPipe = VPHAL_OUTPUT_PIPE_MODE_COMP;
        pRenderPassData->bCompNeeded = true;
    }
And VEBOX is needed if bVeboxBypass is false.
    // Vebox can be bypassed if no feature is needed
    if (!(pRenderData->bDenoise          ||
          pRenderData->bDeinterlace      ||
          pRenderData->bIECP             ||
          pRenderData->bHdr3DLut         ||
          IS_VPHAL_OUTPUT_PIPE_VEBOX(pRenderData)))
    {
        pRenderData->bVeboxBypass = true;
    }
-------------------------------------------------------------------------------------------------------
    case VA_FILTER_SCALING_DEFAULT:
    case VA_FILTER_SCALING_NL_ANAMORPHIC:
    default:
        if (pVpHalSrcSurf->SurfType == SURF_IN_PRIMARY)
        {
            pVpHalSrcSurf->ScalingMode       = VPHAL_SCALING_AVS;
            pVpHalSrcSurf->ScalingPreference = VPHAL_SCALING_PREFER_SFC;
        }

FurongZhang, 2019-05-08:
Let's use this structure to differentiate SFC and AVS(EU)
/** \brief Video Processing Mode. */ 
typedef enum _VAProcMode { 
     /** 
      * \brief Default Mode. 
      * In this mode, pipeline is decided in driver to the appropriate mode. 
      * e.g. a mode that's a balance between power and performance. 
      */ 
     VAProcDefaultMode = 0, 
     /** 
      * \brief Power Saving Mode. 
      * In this mode, pipeline is optimized for power saving. 
      */ 
     VAProcPowerSavingMode, 
     /** 
      * \brief Performance Mode. 
      * In this mode, pipeline is optimized for performance. 
      */ 
     VAProcPerformanceMode 
 } VAProcMode;

FurongZhang, 2019-06-11:
https://github.com/intel/media-driver/pull/641

FurongZhang, 2019-06-11:
@lizhong1008, can you please help to review the code? Thanks a lot in advance!

FurongZhang, 2019-06-11:
If no further concern, we will merge the code and close this issue.

closed 2020-03-05 by XinfengZhang