intel/media-driver #597: Failed to give correct attribute value of VAConfigAttribRTFormat for VP9 10bits

HeJunyan, 2019-04-15:
MediaLibvaCaps::CreateDecAttributes need to handle the VP9 10 bits case for VAConfigAttribRTFormat attribute. vaGetConfigAttributes API query VAConfigAttribRTFormat just get VA_RT_FORMAT_YUV420 for 10bits VP9. So APP failed to create VA_RT_FORMAT_YUV420_10BPP kind surface and cause a decoder setting error.

wangyan-intel, 2019-04-17:
@HeJunyan 
What profile did you try? For FtrIntelVP9VLDProfile2Decoding10bit420, the VA_RT_FORMAT_YUV420_10BPP has been added like the following. We may need add 10bit YUV444 RT format for the other profiles.
https://github.com/intel/media-driver/blob/fb20edf05ba5a4ec4a1bb0b47aed883cb939c44f/media_driver/linux/common/ddi/media_libva_caps.cpp#L882

feiwan1, 2019-04-17:
Hi @wangyan-intel , If possible, could you add all RT formats here for all VP9 profiles base on VP9 profiles definition? Otherwise it will still block 422/444 8/10bit decode feature that we will develop soon.
https://www.webmproject.org/vp9/profiles/

HeJunyan, 2019-04-19:
I verify that, 
On Gen 9.5, FtrIntelVP9VLDProfile2Decoding10bit420 is OK,
On Gen11, no that support and still need
 https://github.com/intel/media-driver/pull/605

HeJunyan, 2019-04-25:
#605 works for me on my icelake platform, thanks

PR #606 (2019-05-27): Add RT formats for VP9 profile 0~3.
Fixes #597.
commit 73e18038 (2019-05-27): Add RT formats for VP9 profile 0~3.

Fixes #597.
closed 2019-04-29 by HeJunyan