intel/media-driver #611: [FFmpeg-vaapi][KBL][max_frame_size] memory leak

fulinjie, 2019-04-24:
apply patch to enable max_frame_size for vaapi:
https://github.com/intel-media-ci/ffmpeg/pull/36/commits/bd592e74a98377483d7ccd40ada459acac3815cd

run cmdline:
valgrind --tool=memcheck --leak-check=full --log-file=webm_log.txt ffmpeg_g -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -s:v 352x288 -i ./MOBILE_352x288_30_orig_01.yuv -vf format=nv12,hwupload -c:v h264_vaapi -profile:v main -b:v 5M -g 30 -bf 3 -max_frame_size 40000 -vframes 100 -y ./vaapi_mobile_max_frame_size.h264

Memory leak info:
==19386== HEAP SUMMARY:
==19386==     in use at exit: 24,481 bytes in 121 blocks
==19386==   total heap usage: 25,224 allocs, 25,103 frees, 48,604,956 bytes allocated
==19386== 
==19386== 4 bytes in 1 blocks are definitely lost in loss record 2 of 19
==19386==    at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==19386==    by 0x1445B824: MOS_AllocAndZeroMemoryUtils (mos_utilities.c:3600)
==19386==    by 0x1498AE37: DdiEncodeAvc::ParseMiscParamMultiPassFrameSize(void*) (media_ddi_encode_avc.cpp:364)
==19386==    by 0x14991A76: DdiEncodeAvc::ParseMiscParams(void*) (media_ddi_encode_avc.cpp:1930)
==19386==    by 0x1498DB27: DdiEncodeAvc::RenderPicture(VADriverContext*, unsigned int, unsigned int*, int) (media_ddi_encode_avc.cpp:940)
==19386==    by 0x14963968: DdiEncode_RenderPicture(VADriverContext*, unsigned int, unsigned int*, int) (media_libva_encoder.cpp:568)
==19386==    by 0x149CA604: DdiMedia_RenderPicture(VADriverContext*, unsigned int, unsigned int*, int) (media_libva.cpp:3209)
==19386==    by 0x6536EA8: vaapi_encode_issue (vaapi_encode.c:432)
==19386==    by 0x6538B75: ff_vaapi_encode_receive_packet (vaapi_encode.c:1007)
==19386==    by 0x5EC159E: avcodec_receive_packet (encode.c:427)
==19386==    by 0x13632A: flush_encoders (ffmpeg.c:1934)
==19386==    by 0x141358: transcode (ffmpeg.c:4712)
==19386== 
==19386== 396 bytes in 99 blocks are definitely lost in loss record 13 of 19
==19386==    at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==19386==    by 0x1445B824: MOS_AllocAndZeroMemoryUtils (mos_utilities.c:3600)
==19386==    by 0x1498AE37: DdiEncodeAvc::ParseMiscParamMultiPassFrameSize(void*) (media_ddi_encode_avc.cpp:364)
==19386==    by 0x14991A76: DdiEncodeAvc::ParseMiscParams(void*) (media_ddi_encode_avc.cpp:1930)
==19386==    by 0x1498DB27: DdiEncodeAvc::RenderPicture(VADriverContext*, unsigned int, unsigned int*, int) (media_ddi_encode_avc.cpp:940)
==19386==    by 0x14963968: DdiEncode_RenderPicture(VADriverContext*, unsigned int, unsigned int*, int) (media_libva_encoder.cpp:568)
==19386==    by 0x149CA604: DdiMedia_RenderPicture(VADriverContext*, unsigned int, unsigned int*, int) (media_libva.cpp:3209)
==19386==    by 0x6536EA8: vaapi_encode_issue (vaapi_encode.c:432)
==19386==    by 0x6538B75: ff_vaapi_encode_receive_packet (vaapi_encode.c:1007)
==19386==    by 0x5EC159E: avcodec_receive_packet (encode.c:427)
==19386==    by 0x13328C: do_video_out (ffmpeg.c:1294)
==19386==    by 0x134348: reap_filters (ffmpeg.c:1504)
==19386== 
==19386== LEAK SUMMARY:
==19386==    definitely lost: 400 bytes in 100 blocks
==19386==    indirectly lost: 0 bytes in 0 blocks
==19386==      possibly lost: 0 bytes in 0 blocks
==19386==    still reachable: 24,081 bytes in 21 blocks
==19386==         suppressed: 0 bytes in 0 blocks
==19386== Reachable blocks (those to which a pointer was found) are not shown.
==19386== To see them, rerun with: --leak-check=full --show-leak-kinds=all

wangyan42164, 2019-05-09:
@fulinjie  Could you please try #619? Thanks.

fulinjie, 2019-05-10:
Thanks @wangyan42164 , after applying the PR, got the dumped log:

HEAP SUMMARY:
==1132==     in use at exit: 24,477 bytes in 120 blocks
==1132==   total heap usage: 25,224 allocs, 25,104 frees, 48,604,956 bytes allocated
==1132== 
==1132== 396 bytes in 99 blocks are definitely lost in loss record 12 of 18
==1132==    at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==1132==    by 0x14464824: MOS_AllocAndZeroMemoryUtils (mos_utilities.c:3600)
==1132==    by 0x14993E99: DdiEncodeAvc::ParseMiscParamMultiPassFrameSize(void*) (media_ddi_encode_avc.cpp:371)
==1132==    by 0x1499AAD8: DdiEncodeAvc::ParseMiscParams(void*) (media_ddi_encode_avc.cpp:1937)
==1132==    by 0x14996B89: DdiEncodeAvc::RenderPicture(VADriverContext*, unsigned int, unsigned int*, int) (media_ddi_encode_avc.cpp:947)
==1132==    by 0x1496C968: DdiEncode_RenderPicture(VADriverContext*, unsigned int, unsigned int*, int) (media_libva_encoder.cpp:568)
==1132==    by 0x149D3666: DdiMedia_RenderPicture(VADriverContext*, unsigned int, unsigned int*, int) (media_libva.cpp:3209)
==1132==    by 0x653F93A: vaapi_encode_issue (vaapi_encode.c:432)
==1132==    by 0x6541607: ff_vaapi_encode_receive_packet (vaapi_encode.c:1007)
==1132==    by 0x5EC8E5E: avcodec_receive_packet (encode.c:427)
==1132==    by 0x13328C: do_video_out (ffmpeg.c:1294)
==1132==    by 0x134348: reap_filters (ffmpeg.c:1504)
==1132== 
==1132== LEAK SUMMARY:
==1132==    definitely lost: 396 bytes in 99 blocks
==1132==    indirectly lost: 0 bytes in 0 blocks
==1132==      possibly lost: 0 bytes in 0 blocks
==1132==    still reachable: 24,081 bytes in 21 blocks
==1132==         suppressed: 0 bytes in 0 blocks
==1132== Reachable blocks (those to which a pointer was found) are not shown.
==1132== To see them, rerun with: --leak-check=full --show-leak-kinds=all

The first leaking issue is solved, but the second still occurs.

wangyan42164, 2019-05-10:
@fulinjie Thanks for your test. It seems only reduce 4 bytes leak based your latest log. I update my PR for free after encoding every time. Could you please try it again? Thanks.

fulinjie, 2019-05-21:
@wangyan42164 Sorry for replying late, I had tested it early but didn't update the result until now.

==15144== HEAP SUMMARY:
==15144==     in use at exit: 24,081 bytes in 21 blocks
==15144==   total heap usage: 25,224 allocs, 25,203 frees, 48,604,956 bytes allocated
==15144==
==15144== LEAK SUMMARY:
==15144==    definitely lost: 0 bytes in 0 blocks
==15144==    indirectly lost: 0 bytes in 0 blocks
==15144==      possibly lost: 0 bytes in 0 blocks
==15144==    still reachable: 24,081 bytes in 21 blocks
==15144==         suppressed: 0 bytes in 0 blocks
==15144== Reachable blocks (those to which a pointer was found) are not shown.
==15144== To see them, rerun with: --leak-check=full --show-leak-kinds=all

After applying the updated PR, all the memory leaks are gone.
Thanks a lot.

PR #619 (2019-05-21): Add free logic of picParams->pDeltaQp for avoiding memory leak.
Fixes #611.
commit b672825c (2019-05-21): Add free logic of picParams->pDeltaQp for avoiding memory leak.

Fixes #611.
closed 2019-05-21 by intel-mediadev