intel/media-driver #612: [avc vdenc] multislice encode file have garbage

zhuqingliang, 2019-04-24:
bug describe
NOT  lowpower mode , multislice encode  is good 
enable lowpower mode , the h264 file have garbage ,  both in ffmpeg and gstreamer.
1. environment
Platform: KBL
Platform: ICL
Arch: x86_64
Kernel: 5.0.0
Libva: c98b06d2b8c00dc4df628488b672711b3f0eb118
iHD_driver: 5366de39a3f3b7f256b540bfa0ba088133315610
libva-utils: 375e4eaae3377c1806e83874f9fa9b79b1f225b1
ffmpeg-vaapi: 9dece050ef01c70df1ac74a04da3548b3c0d79a9
2 .
ffmpeg 
step 1 : 
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -pix_fmt yuv420p -s:v 720x480 -i  720x480p.yuv -vf 'format=nv12,hwupload' -c:v h264_vaapi -profile:v high -rc_mode CQP -g 30 -qp 10 -quality 4 -slices 3 -low_power 1 -vframes 150 -y 720x480p.264

step 2:  
ffmpeg -i  720x480p.264 -pix_fmt yuv420p -f rawvideo  -vframes 1 -vsync passthrough  -y dump.yuv 
dump.yuv have garbage :
error message:
[h264 @ 0x1e63080] decode_slice_header error
[h264 @ 0x1e63080] slice type 32 too large at 1215
[h264 @ 0x1e63080] decode_slice_header error
[h264 @ 0x1e63080] slice type 32 too large at 1260
[h264 @ 0x1e63080] decode_slice_header error
[h264 @ 0x1e63080] slice type 32 too large at 1305

uartie, 2019-04-24:
This is a duplicate of #389

zhuqingliang, 2019-04-25:
reduplicate , close .

closed 2019-04-25 by zhuqingliang