intel/media-driver #613: AVC low power encoding fails on ubuntu 19.04 with intel-media-va-driver

dvrogozh, 2019-04-27:
Derived from https://github.com/Intel-Media-SDK/MediaSDK/issues/1398.

AVC low power encoding fails under Ubuntu 19.04 under `intel-media-va-driver ` because of 2 reasons:
1. Bug in mediasdk which can be addressed by for example Intel-Media-SDK/MediaSDK#1399
2. Becuse intel/media-driver@26d7d50 commit is missed in https://github.com/intel/media-driver/releases/tag/intel-media-18.4.1 release which is used by Ubuntu 19.04

**To reproduce the issue:**

Install Ubuntu 19.04 and mediasdk coming from the OS:
```
sudo apt-get install libmfx1 libmfx-tools
sudo apt-get install libva-drm2 libva-x11-2
sudo apt-get install intel-media-va-driver
```

And run the following sample encode command line to encode via AVC Low Power encoder:
```
export LIBVA_DRIVER_NAME=iHD

# just to prepare some YUV input file:
wget https://fate-suite.libav.org/h264-conformance/AUD_MW_E.264
/usr/share/mfx/samples/sample_decode h264 -i AUD_MW_E.264 -o AUD_MW_E.yuv

# and now encode:
sample_encode: /build/intel-mediasdk-AzMSof/intel-mediasdk-18.4.1/_studio/mfx_lib/shared/src/mfx_h264_encode_vaapi.cpp:1422: virtual mfxStatus MfxHwH264Encode::VAAPIEncoder::CreateAuxilliaryDevice(VideoCORE*, GUID, mfxU32, mfxU32, bool): Assertion `0x00000000 == vaSts' failed.

Aborted (core dumped)
```

dvrogozh, 2019-04-27:
@tjaalton: can you, please, consider to update media driver with 19.1 release to pull in 26d7d50? or we need to cherry-pick the change to 18.4 which you currently use in Ubuntu 19.04?

tjaalton, 2019-04-29:
hi, I probably need to cherry-pick the fix..

dvrogozh, 2019-10-18:
Fixed w/ media driver package update in Ubuntu 19.10. Please, consider updating Ubuntu regular release from 19.04 to 19.10 if you are affected by this issue.

closed 2019-10-18 by dvrogozh