intel/media-driver #616: [gstreamer-vaapi][encode][Jpeg] jpeg file show noise by using UYVY input format.

feiwan1, 2019-04-30:
When using UYVY format as the input for jpeg encode, the output file show noise.
(This need the patch #615 )

cmdline:
gst-launch-1.0 videotestsrc num-buffers=1 ! video/x-raw,format=UYVY ! vaapijpegenc ! jpegparse ! filesink location=out.uyvy.jpeg

And I used jpegenc from libva-utils got the same result:
jpegenc 320 240 in.uyvy out.jpeg 2 50

Ps, when using YUY2, the output looks good.

Noise picture:
[image]

Expected picture:
[image]

wangyan-intel, 2019-05-09:
@feiwan1 I tried your command with #615 but no this issue on KBL. What platform do you run it on? My tried commit is 54a198. Thanks.

feiwan1, 2019-05-09:
I am using SKL. @wangyan-intel

dnagasun, 2019-06-10:
I tried the command with #615 in EHL Simics, can see this issue. 
it fails with UYVY, but looks good when tried with YUY2.

wangyan-intel, 2019-06-19:
@feiwan1 Could you please try #648 ? Thanks.

feiwan1, 2019-06-19:
@wangyan-intel , with #615 and this patch, the issue is gone on SKL. Thanks.

wangyan42164, 2019-06-20:
For ICL platform, also need #658.

PR #648 (2019-06-26): UYVY format should use Y tiling.
Fixes #616.
commit e65ab913 (2019-06-26): UYVY format should use Y tiling.

Fixes #616.
closed 2019-06-26 by intel-mediadev