intel/media-driver #618: [ffmpeg-vaapi][HEVC encode][ICL][low power][ICQ support]

fulinjie, 2019-05-08:
I'm enabling low power encode for ffmpeg hevc_vaapi.
patch:
https://github.com/fulinjie/ffmpeg/commit/3c7b3eea0070d71d752a53db6b94520be8c6a29e

cmdline:
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -s:v 1920x1080 -pix_fmt yuv420p -i 1080p_blue_sky.yuv -vf format=nv12,hwupload -c:v hevc_vaapi -low_power 1 -global_quality 25 -vframes 5 -y ./vaapi.h265

Known limitation is:
- media-driver require CTU size must be set as 64x64 at low power mode
- cu_qp_delta_enabled_flag should be enabled for low power mode
- low delay B to replace P frame and same L0/L1

https://github.com/intel/libva/pull/220#issuecomment-481180841

When enabling h265_vaapi VDENC, i can see CBR/VBR/CQP mode can work,but the default RC mode(ICQ) can not be supported for this:
[CODEC]:CRITICAL - SetDmemHuCBrcInitReset:3364: VBV BufSize should not be 0 for BRC case

The reason is ICQ is taken as m_brcEnabled but without vbvbuf_size :
https://github.com/intel/media-driver/blob/abb20bb6d3c433fbfef3978b273563036a15380a/media_driver/agnostic/gen11/codec/hal/codechal_vdenc_hevc_g11.cpp#L3362

I think there is no need for setting vbv bufsize or bits_per_second in ICQ mode.
Also checking some other codecs:
I can see h264_vaapi doesn't support ICQ mode:
[h264_vaapi @ 0x135cd40] Driver supports RC modes CQP, CBR, VBR, QVBR.
[h264_vaapi @ 0x135cd40] Driver does not support ICQ RC mode.
[h264_vaapi @ 0x135cd40] RC mode: CQP.

hevc_qsv doesn't support ICQ in low_power mode.

My question is, 
1. Is there any other constraint for enabling low power encode for hevc? 
2. Is ICQ mode supported in low_power mode for ICL?
If so, this seems to be an issue and need to be fixed.
If not, ICQ mode shouled be disabled in ffmpeg level.
3. Will this query be supported soon, to add in ffmpeg-vaapi?

Xiaogangli-intel, 2019-05-21:
Hi,
    ICQ mode is not supported in low power mode on ICL, it should be disabled in app level.

fulinjie, 2019-05-28:
FFmpeg will try the allowed RC modes according to query result in vaGetConfigAttributes.
Disable this in hard code is not acceptable in the community.

IMHO,If not supported, vaGetConfigAttributes shouldn't have reported the support for ICQ mode for VAEntrypointEncSliceLP.

wangyan42164, 2019-07-16:
@fulinjie Could you please try #683? Thanks.

fulinjie, 2019-07-30:
HI @wangyan42164 ,
after applying this pr, ICQ mode is still quried as supported mode for ICL.

[hevc_vaapi @ 0x1a94480] Input surface format is nv12.
[hevc_vaapi @ 0x1a94480] Using VAAPI profile VAProfileHEVCMain (17).
[hevc_vaapi @ 0x1a94480] Using VAAPI entrypoint VAEntrypointEncSliceLP (8).
[hevc_vaapi @ 0x1a94480] Using VAAPI render target format YUV420 (0x1).
[hevc_vaapi @ 0x1a94480] Driver supports RC modes CQP, CBR, VBR, ICQ, QVBR.
[hevc_vaapi @ 0x1a94480] No quality level set; using default (25).
[hevc_vaapi @ 0x1a94480] RC mode: ICQ.
[hevc_vaapi @ 0x1a94480] RC quality: 25.
[hevc_vaapi @ 0x1a94480] RC framerate: 25/1 (25.00 fps).
[hevc_vaapi @ 0x1a94480] Using intra and P-frames (supported references: 3 / 0).
[hevc_vaapi @ 0x1a94480] All wanted packed headers available (wanted 0xd, found 0x1f).
[hevc_vaapi @ 0x1a94480] Failed to create encode pipeline configuration: 10 (attribute not supported).
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
[AVIOContext @ 0x1a95400] Statistics: 0 seeks, 0 writeouts
[AVIOContext @ 0x1a8fdc0] Statistics: 3110400 bytes read, 0 seeks
Conversion failed!

wangyan-intel, 2020-06-01:
@fulinjie 
I think current driver has disabled ICQ mode for VDENC:
https://github.com/intel/media-driver/blob/master/media_driver/linux/common/ddi/media_libva_caps.cpp#L663
Could you please check it?
If this issue has been fixed, could you please close it?
Thanks.

PR #954 (2020-06-08): Disable ICQ mode for HEVC VDENC.
ICQ mode isn't supported by HEVC VDENC.
Fixes #618.
commit 6545b4c1 (2020-06-08): Disable ICQ mode for HEVC VDENC.

ICQ mode isn't supported by HEVC VDENC.
Fixes #618.
closed 2020-06-08 by intel-mediadev