intel/media-driver #620: Do not map linear surface with no cache.

xuguangxin, 2019-05-09:
The driver [always ](https://github.com/intel/media-driver/blob/8c96c595e753c7f7cb8b213703a53a2ddd6034c7/media_driver/linux/common/ddi/media_libva_util.cpp#L1000)uses mapgtt on the atom.  No matter it’s tiled or linear surface. This logic does not exist on [i965](https://github.com/intel/intel-vaapi-driver/blob/72f10f16f7e5767492acde130eeeaa598d26a3a6/src/i965_drv_video.c#L3065).
The logic will disable cache for linear surface and slow down the cpu access. 
If we have this issue, two important and typical use cases are not possible：
1. decode output -> scale and csc to i420 surface -> send i420 surface directly to sw encoder.
2. decode output -> scale and csc to RGB surface -> use RGB surface directly for cpu based post-processing.

Please help fix it.
thanks

FurongZhang, 2019-05-09:
I will be OOO ww20. Will follow up this issue ww21.

xuguangxin, 2019-05-10:
Thanks, Furong for quick response

FurongZhang, 2019-05-27:
Sorry for the late response because of taking annual leave. I will check this issue this week.

FurongZhang, 2019-06-13:
@xuguangxin , I filed pull request for this issue. Please help to review that, thanks a lot in advance!
https://github.com/intel/media-driver/pull/646

xuguangxin, 2019-06-14:
thanks @FurongZhang , @fulinjie could you help check it benefits.
_decode output -> scale and csc to i420 surface -> send i420 surface directly to x264_
or not?
thanks

fulinjie, 2019-06-17:
Trid above pipeline but failed with cmdline:
ffmpeg -hwaccel qsv -c:v h264_qsv -i 9-\ 4K-\ Nature-\ H264-\ SDR_60fps.mp4 -vf "scale_qsv=format=yuv420p,hwdownload,format=yuv420p" -c:v libx264 -f h264 /dev/null

And I tried HW decode + CSC to rgb32 pipeline on APL, the performance is similar:
ffmpeg -hwaccel qsv -c:v h264_qsv -i 9-\ 4K-\ Nature-\ H264-\ SDR_60fps.mp4 -vf "scale_qsv=format=rgb32,hwdownload,format=rgb32" -f rawvideo -pix_fmt rgb32 /dev/null

frame=   41 fps=3.9 q=-0.0 Lsize= 1328400kB time=00:00:00.68 bitrate=15909330.9kbits/s speed=0.0652x

With gpu_copy, it can reach up to 30 fps.

xuguangxin, 2019-06-19:
Hi @FurongZhang , 
I double confirmed @fulinjie 's finding. 
The iHD driver uses mos_bo_map map linear surface, but the performance is still not good.

you can use libyami to reproduce the issue.  the command is
`$yamidecode -i test.264   -m 0`

use iHD driver with your patch, we only get about 10fps for a 1080P video.
use i965 driver, it can reach 100+fps.

Is anything we missed here?

thanks

xuguangxin, 2019-06-24:
@FurongZhang , any feedback on this?
thanks

FurongZhang, 2019-06-24:
I will check this function mos_bo_map and why it does not take effect as expected.

xuguangxin, 2019-06-28:
this @FurongZhang , waiting for your feedback

xuguangxin, 2019-10-24:
@XinfengZhang , any progress on this?

thanks

FurongZhang, 2022-11-18:
Please use this patch https://github.com/intel/media-driver/pull/646/files.

closed 2022-11-18 by FurongZhang