intel/media-driver #621: With LIBVA_DRIVER_NAME=iHD variable set in /etc/enviroment only h 264 works but without it Vp 9 works also

freefreeno, 2019-05-12:
Sorry for the length of this but this is proof and I want to back up my claims. I am using Arch Linux with Intel HD 620 and Mesa and the this Intel-media-driver installed. No other drivers.
With the LIBVA_DRIVER_NAME=iHD set in /etc/environment the vainfo command shows this:
[image]

Also I have these settings in my graphics:

[image]

 When I try to play a Vp 9 video it just loads and never plays.

[image]

[image]

But h 264 works fine

[image]

[image]

Ok. Now with if I remove the variable from the /etc/environment I can't use the vainfo command because it shows error because as far as I know you have to have the variable.

[image]

But even though the vainfo command shows error all my videos start back playing with both Vp9 and h 264. This is the exact same videos as above only this time without the variable.
This is the Vp9 youtube that would not play at all with variable set in Arch linux.

[image]

[image]

And I can still play my h264 content just as I could with variable set.
Same h264 video as above but this time without variable.

[image]

[image]

Now that is the problem. All the guides including Arch wiki say to set that variable and my system will not work with the variable set and and if you add the path which is LIBVA_DRIVERS_PATH=/usr/lib/dri/iHD_drv_video.so the vainfo command will not even work then even with both path and the normal LIBVA_DRIVER_NAME=iHD variable set. i said that because I know someone is gonna say to set the path also but I have tried.

wangyan-intel, 2019-05-13:
@freefreeno After remove LIBVA_DRIVER_NAME=iHD, chrome browser should fall back to SW decoding. For VP9, I can see VpxVideoDecoder from your screenshot of chrome://media-internals.  I am not sure which commit of iHD driver is used. Could you please try the latest commit? Recently, some VP9 profile patches are merged. Thanks.

freefreeno, 2019-05-13:
I am using from Arch AUR and it is the 19.1.0-1 so i think it is the latest.

freefreeno, 2019-05-13:
Chromium is not using the software rendering after the variable is removed I dont't think. As you can see my videos are playing h264 and Vp9 videos with the variable removed but with the variable I can only play h 264. I don't even understand how the h 264 and Vp9 videos are being played after the variable is removed.

wangyan-intel, 2019-05-14:
@freefreeno I mean software decoder not software rendering.  VpxVideoDecoder from your chrome://media-internals should be software decoder. And MojoVideoDecoder is GPU accelerated which will use iHD driver. You can check chromium code to confirm it. 
H264 is similar. When iHD is enabled, chrome browser will use its GPU wrapper to call iHD driver for decoding, otherwise use software decoder.
Just current MojoVideoDecoder/iHD driver of Arch Linux can decode H264 but cannot decode VP9 rightly.  But software video decoder can decode both H264 and VP9 rightly.
Arch Linux may not use the latest commit of iHD driver like commit b88f49f. Could you please check it? Or you can build iHD driver manually to replace it. 
Thanks.

freefreeno, 2019-05-14:
I would Like to show you something. Just to see if it would change things I removed the driver and installed the older libva-intel and when I type the vainfo command I see that now I have the slice next to Vp9 and with you alls intel-media-driver I did not have that.

[image]

Ok so what needs to happen is the software decoder should pick up what the driver can't decode but that is not what is happening. If I set the variable yes I can hardware decode everything but Vp9 but the software decoder does not pick up the slack. The software decoder does not work at all if you set the variable. It is either hardware decoded or the video does not play at all. So I can play all videos hardware decoded but the Vp9 and  I can't play that all. But you will see above that the regular libva driver shows i can decode Vp9 but the same thing happens and the Vp9 hardware decoded videos fail to even play at all with i965 variable set.

wangyan-intel, 2019-05-15:
1.  intel-media-driver (i965) driver is different with iHD driver. 
2. In fact, I enabled VP9 8K video playback on Chrome OS browser with iHD driver on KBL platform recently. It just need my merged patch baedcc3 for 8K. I am not sure what is your video resolution. If it is not larger than than 4K, it should works without this patch. I can see the obvious difference between SW and HW accelerated decoder. It works fine on Chrome OS browser with iHD driver.
Your issue should come from Arch Linux configuration or older iHD driver version.

freefreeno, 2019-05-15:
I compiled this driver yesterday and the same problem is still there. Mesa has been updated this morning. The hardware decoded Vp9 videos do not work on the old libva-intel and your intel-medi-driver. It did work a couple weeks back. Your driver doesn't even show to be capable to run Vp9 hardware decode on my system and it did a week or so ago. I have done everything possibel including building and installing. it just doesn't work on my Intel HD 620 and what really sucks is once you set that variable then when a video fails to hardware decode well it just doesn't play AT ALL so I have to go remove the variable and then it works with software. plain and simple it should fall back to software decode if hardware fails. Everything works but Vp9 with hardware accel but it doesn't automatically fall back to software decoder.

wangyan-intel, 2019-05-30:
@freefreeno Do you try it on Ubuntu/Fedora with the chromium upstream? Ubuntu/Fedora may be easier to debug and locate this issue. Thanks.

freefreeno, 2019-05-30:
Uh NO. How would an operating system that I have never used be easier to debug. I have everything set just like every wiki and everyone says and there is more than then environment variable going on. Me and everyone you see listed above says there Vp9 does not play any longer and that goes for intel-vaapi or intel-media driver. I can remove all the variables with the same results. I just wanted to know why I can not set the path variable and it work as it is supposed to. I can put the variable name in /etc/environment but not the path because when I do all the hardware acceleration stops working.

wangyan-intel, 2019-05-30:
@freefreeno I have not ArchLinux + Chromium browser to test currently. And I cannot find this issue on Ubuntu/Fedora.

XinfengZhang, 2019-05-30:
@freefreeno maybe you can enable libva trace and upload the logs also try debug version media driver to tell us the root cause

freefreeno, 2019-05-30:
Just an overview. I have all the latest packages from Arch Linux. When I did a clean install about two weeks ago I went thru everything again. Enabled early kms. Enabled huc and guc and can confirm kms and huc and guc is loaded.  I have intel-media-driver, xf86-video-fbdev and  libva-utils installed. I can set the variable for the name like so-LIBVA_DRIVER_NAME=iHD
BUT if I set the variable for the path which is LIBVA_DRIVERS_PATH=/usr/lib/dri/iHD_drv_video.so
then the vainfo command starts throwing error 1. So vainfo works without the path set but you do not see the part where it says trying to open the driver and I can not hardware decode Vp9. This is what vainfo says without PATH variable set. Even vainfo shows I am unable to decode Vp9. I have been stuck like this for a while.
[image]

wangyan-intel, 2019-05-30:
@freefreeno I am not sure if there is also i965 media driver at your platform. If yes, could you please remove it fully? Thanks.

freefreeno, 2019-05-31:
Yes we have the intel-vaapi driver in ther AUR and I have never installed it since I did my clean install 2 weeks back. I did try it and IT DID JUST LIKE THE INTEL MEDIA DRIVER. Same exact thing. I could set the name variable but not the path. Makes me wonder if it isn't a Arch Linux problem or not a problem but maybe something they are doing differently. I just want my stuff to work and I may not be the smartest person in the world but I am doing everything that is supposed to be done. I have also never had xf86-video-intel installed on this setup either so there has been nothing but mesa and I am on version  19.0.5-1.

XinfengZhang, 2019-05-31:
hi @freefreeno ,  1.  should set LIBVA_DRIVERS_PATH=/usr/lib/dri/, this variable should only include the path name exclude the driver so filename , vainfo should works with  LIBVA_DRIVER_NAME=iHD & LIBVA_DRIVERS_PATH=/usr/lib/dri/ . 2. I still prefer to get the libva log or media driver log. simply export LIBVA_TRACE=folder name. you can get the libva traces. we can help check the call sequence. thanks.

freefreeno, 2019-05-31:
NO IT DOES  NOT AND I HAVE EXPLAINED THAT. Vainfo only works if I set it like this.
LIBVA_DRIVER_NAME=iHD

If I set it like this it shows error 1.
LIBVA_DRIVERS_PATH=/usr/lib/dri/iHD_drv_video.so
LIBVA_DRIVER_NAME=iHD

BUT I can set it like this but IT WILL NOT SAY OPENING THE DRIVER in the vainfo command.
It should say opening driver in the vainfo command and the reason it does not is because I CAN NOT SET THE PATH. I know what I am doing a little bit. Something is wrong.

LIBVA_DRIVERS_PATH=/usr/lib/dri/
LIBVA_DRIVER_NAME=iHD

I am sorry but I do not know how much clearer I can be.

uartie, 2019-05-31:
@freefreeno using ALL CAPS is _generally_ viewed/interpreted as **SHOUTING**.  If you need to emphasize something, please use other means like, **bold**, *italics*, etc...  Thanks.

freefreeno, 2019-05-31:
I am not shouting but ok.

uartie, 2019-06-03:
> I am not shouting but ok. lol

No worries.  It's just a netiquette thing and wouldn't want your posts to be taken the wrong way ;-)

freefreeno, 2019-06-03:
Sorry. If anything I am aggravated. I have posted to Arch site this week with no help there yet. Can someone please tell me if  the old intel driver can be ran with the intel-media-driver?? I found some info on Arch wiki that says that if you want to take advantage of huc and guc and other things as well like fbc then that requires the old regular intel driver. I did not know that and I am not reading in between the lines to come up with that because it plainly say it in either the intel graphics section or the hardware acceleration section.Can someone to confirm or not confirm if the intel-media-driver can ran with xf86-video-intel?

alfinbi, 2019-06-27:
I also have same issue using arch linux and intel HD 630, it used to work few months before

freefreeno, 2019-06-29:
@alfinbi  I have still not been able to resolve my issues and now I don't even think I can play any videos. I believe the issues is with the new chromium's. If you will install version 74.0.3729.169-2-x86_64 then I believe you will start back working then. Go to the aur and look at the comments there. It is known that the new versions break or hardware accel on Intel graphics but they are leaving it up to us to block the new updates.

Hergeirs, 2019-11-14:
I have a similar issue. Use Archlinux as well. No crash here. Just unable to play VP9 or VP8 with hw-acceleration enabled in epiphany browser. 

What I've noticed is that the problem only seems to happen for me in applications that use gstreamer. I've installed gstreamer-vaapi and when I set LIBVA_DRIVER_NAME=iHD along with GST_VAAPI_ALL_DRIVERS=1 (because gstreamer won't actually use iHD if this isn't set. see: https://wiki.archlinux.org/index.php/GStreamer#Hardware_video_acceleration)

When using mpv which uses ffmpeg as backend vaapi works as expected for both VP8 and VP9.
I'm just wondering where the issue lies. Am I right to assume that this would indicate that the issue is with gstreamer-vaapi plugin?

tmathmeyer, 2020-08-13:
I've also started experiencing this issue on debian with the iHD 20.1.1 driver version.

we're working on shipping HW acceleration in chrome by default for linux here: https://chromium-review.googlesource.com/c/chromium/src/+/2288633

you'll notice that we used a workaround to ask chrome to switch to the i965 driver, if available. Obviously, we'd like to not have this workaround, but it seems that unless the crash goes away, we're stuck with it. I believe we are getting the same crash as described in issue #818 by calling VaPutSurface.

tmathmeyer, 2020-08-14:
also, as a note to @freefreeno
the reason your videos play when not using the flag is because they are being played using a software decoder. You can see that the decoder name is VpxVideoDecoder. This is one of the software decoders in chromium, the other common one being FfmpegVideoDecoder (which you will see for h264 videos)

DecryptingVideoDecoder is using widevine, which can punt the decoding issue off to software or hardware, depending on the license restriction from the video hosting service.

Ideally you'd be seeing MojoVideoDecoder, which is a hardware decoder running in a separate thread with access to GPU resources. This _should_ be coming by default to chromium / chrome soon.

Jexu, 2021-11-23:
We will close this issue since it exists for long time. Please feel free to re-open it again if you still meet similar issue. Thanks a lot.

closed 2021-11-23 by Jexu