intel/media-driver #624: [FFmpeg-vaapi][ICL][VP9][VDENC]is there any BKM or constraints for vp9 vdenc?

fulinjie, 2019-05-16:
Any constraints or BKMs for enabling vp9 VDENC? 
Like hevc VDENC does in https://github.com/intel/media-driver/issues/618

fulinjie, 2019-05-16:
Currently, i see only VDENC is allowed for VP9 encode (VAEntrypointEncSliceLP),

With cmdline:
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -s:v 1920x1080 -pix_fmt yuv420p -i 1080p_blue_sky.yuv -vf format=nv12,hwupload -c:v vp9_vaapi -qp 30 -vframes 100 -y ./vaapi.webm

Error message:
[AVBSFContext @ 0x20976c0] Too many invisible frames
[AVBSFContext @ 0x20976c0] Failed to send packet to filter vp9_superframe for stream 0

And regular P frame is used for encoding currently, is there low delay B needed, too?

fulinjie, 2019-05-28:
@Xiaogangli-intel 
Hi Xiaogang, update some new information for VP9 VDENC.

Error message:
vp9_vaapi @ 0x2039fc0] Input surface format is nv12.
[vp9_vaapi @ 0x2039fc0] Using VAAPI profile VAProfileVP9Profile0 (19).
[vp9_vaapi @ 0x2039fc0] Using VAAPI entrypoint VAEntrypointEncSliceLP (8).
[vp9_vaapi @ 0x2039fc0] Using VAAPI render target format YUV420 (0x1).
[vp9_vaapi @ 0x2039fc0] No quality level set; using default (100).
[vp9_vaapi @ 0x2039fc0] RC mode: CQP.
[vp9_vaapi @ 0x2039fc0] RC quality: 100.
[vp9_vaapi @ 0x2039fc0] RC framerate: 25/1 (25.00 fps).
[vp9_vaapi @ 0x2039fc0] Using intra and P-frames (supported references: 3 / 0).
[vp9_vaapi @ 0x2039fc0] Driver does not support any packed headers (none wanted).
Output #0, ivf, to './vaapi.ivf':
  Metadata:
    encoder         : Lavf58.27.103
    Stream #0:0: Video: vp9 (vp9_vaapi) (Profile 0), 1 reference frame (VP90 / 0x30395056), vaapi_vld, 1920x1080, q=-1--1, 25 fps, 25 tbn, 25 tbc
    Metadata:
      encoder         : Lavc58.52.101 vp9_vaapi
[MOS]:  CRITICAL - MOS_UserFeature_WriteValues_Tbl_ID:6289: Invalid (nullptr) Pointer.
[MOS]:  CRITICAL - MOS_UserFeature_WriteValues_Tbl_ID:6289: Invalid (nullptr) Pointer.
[MOS]:  CRITICAL - MOS_UserFeature_WriteValues_Tbl_ID:6289: Invalid (nullptr) Pointer.
Automatically inserted bitstream filter 'vp9_superframe'; args=''
[AVBSFContext @ 0x2357b80] Too many invisible frames
[AVBSFContext @ 0x2357b80] Failed to send packet to filter vp9_superframe for stream 0
[AVBSFContext @ 0x2357b80] Too many invisible frames
[AVBSFContext @ 0x2357b80] Failed to send packet to filter vp9_superframe for stream 0
[AVBSFContext @ 0x2357b80] Too many invisible frames
[AVBSFContext @ 0x2357b80] Failed to send packet to filter vp9_superframe for stream 0
[AVBSFContext @ 0x2357b80] Too many invisible frames
[AVBSFContext @ 0x2357b80] Failed to send packet to filter vp9_superframe for stream 0

In vaapi_encode_output,
ffmpeg calls vaMapBuffer to map data to buf_list.

https://github.com/FFmpeg/FFmpeg/blob/9b069eb14e07d8faec32f2eef2d12e514290268f/libavcodec/vaapi_encode.c#L508

It seems the returned buf_list has 2 member:
The first buf contains the normal encoded data:
 p *buf
$134 = {size = 110424, bit_offset = 0, status = 0, reserved = 0, buf = 0x7fffe7dd1000, **next = 0x83e9e8**, va_reserved = {0, 0, 0, 0}}

But the next returned buf has size 0 and no data is available.
p *buf
$132 = {**size = 0**, bit_offset = 65537, status = 0, reserved = 0, buf = 0x0, next = 0x0, va_reserved = {0, 0, 0, 0}}

And this zero buffer covers the previous correct encoded data.
That's the root cause for VP9 VDENC failed. 
Break in the loop if buf->size equals zero can simply work around this issue.

Here are the questions:
1.  It seems that VA_CODED_BUF_STATUS_SINGLE_NALU allows driver to map buf for each slice, is VP9 VDENC enabled this feature?
 
2. What does the returned buf with size = 0 stand for ?

3. I guess ffmpeg should be able to cope with the multiple buf situation, but is this something should be fixed in driver currently?

fulinjie, 2019-05-29:
Cope with potential multiple buf situation in ffmpeg level:
https://github.com/intel-media-ci/ffmpeg/pull/44

wangyan42164, 2019-07-02:
@fulinjie Media driver will return 2 buffers in fact. The first buffer is VACodedBufferSegment buffer which includes encoded output. And the second buffer is extra VACodedBufferVP9Status. If ffmpeg needn't it,  it could be ignored. Please refer to the following URL:
https://github.com/intel/media-driver/blob/master/media_driver/linux/common/codec/ddi/media_ddi_encode_vp9.cpp#L1030

wangyan-intel, 2020-03-11:
https://github.com/intel/media-driver/commit/a2ca915e021fbd272e61cb29042fbf8b8107cce5 should fix it.

closed 2020-03-11 by wangyan-intel