intel/media-driver #625: [ffmpeg-vaapi][h264 encode] VBR encode output bitrate disparities between the QSV and VAAPI encoders.

Brainiarc7, 2019-05-19:
Hello there,

Similar to this ticket: https://trac.ffmpeg.org/ticket/7629
I'm observing major issues with VBR encoding's bitrate adherence disparities between h264_qsv and h264_vaapi, made even worse when switching between VAAPI drivers, ie `i965` and `iHD`.
See this ticket for logs and more info: https://trac.ffmpeg.org/ticket/7910
It could be an issue with the `media-driver` package, or failing rate control implementation for VBR with FFmpeg's h264_vaapi encoder.

Brainiarc7, 2019-05-22:
Any updates on this?

wangyan-intel, 2019-06-04:
@Brainiarc7 Could you please share your rv,mkv in  https://trac.ffmpeg.org/ticket/7910? We can try to reproduce it. Thanks.

Brainiarc7, 2019-06-11:
Will do, thanks.

Brainiarc7, 2019-06-11:
@wangyan-intel, here comes:

1. rv.mkv: https://www.dropbox.com/s/7svk3lbujb2hq8d/rv.mkv?dl=1

wangyan-intel, 2019-06-11:
@Brainiarc7 We will check it. Thanks.

wangyan-intel, 2019-06-11:
@Brainiarc7 I found I cannot access dropbox URL. Could you please share it on Google drive? Sorry for this.

Brainiarc7, 2019-06-12:
Hello there, you should be able to fetch it via wget, as shown:

```sh
wget -c -v https://www.dropbox.com/s/7svk3lbujb2hq8d/rv.mkv?dl=1 
```

Brainiarc7, 2019-06-12:
Now on Google drive https://drive.google.com/open?id=1CRdDZZPHdgHVlvNMZnOjRiglyG_b0XqY

wangyan-intel, 2019-06-12:
@Brainiarc7 Very thanks. I get it now.

Brainiarc7, 2019-06-15:
Do post back with updates on this issue, thanks.

wangyan-intel, 2019-06-18:
@Brainiarc7 
Sorry for slow response. 
I debugged your command including ffmpeg h264_vaapi and qsv. I can reproduce your result. 
In general, the result comes from libVA interface limitation which causes -minrate:v invalid in fact. 
The following is the detail:
1. ffmpeg h264_vaapi passes VAEncMiscParameterRateControl structure to media driver for setting rate control parameters. It is mainly defined by bits_per_sencond and target_percentage which are calculated from max rate and target rate. So the min rate in the media driver isn't passed by the structure from ffmpeg and is calculated by bits_per_sencond and target_percentage. For your command, the target rate is 7.5M and target percentage is 50%. So the min rate is 0 and the max rate is 15M. So if you set max rate is 7.5M, you received bit rate will be higher.
2. For QSV/MSDK. I also did some initial debugging. I found it doesn't use VBR mode based on your command. In fact, it uses CQP mode. And I think it will calculate one target bit rate based on your input. It may consider min rate too.  This need double check from MSDK side.
So if we want that ffmpeg h264_vaapi  has the same result with ffmpeg qsv, we may have to modify the logic of ffmpeg encoder.
Thanks.
Yan Wang

wangyan-intel, 2020-05-13:
It should be ffmpeg issue and close it.

closed 2020-05-13 by wangyan-intel

Brainiarc7, 2020-06-26:
What's the resolution?
Has this been resolved upstream, say, in libav/ffmpeg?