intel/media-driver #630: VP9 encode support from Kaby Lake+

EwoutH, 2019-05-21:
The [Readme.md](https://github.com/intel/media-driver/blob/master/README.md) states VP9 encoding will only be supported from Ice Lake forward. The [Readme](https://github.com/intel/intel-vaapi-driver/blob/master/README) and [code](https://github.com/intel/intel-vaapi-driver/blob/master/src/gen9_vp9_encoder.c) over at [intel/intel-vaapi-driver](https://github.com/intel/intel-vaapi-driver) however states that VP9 encoding is supported from Kaby Lake forward.

So:

1. Is encoding VP9 with Kaby Lake possible?
2. Is encoding VP9 with Kaby Lake implemented in this repo?

XinfengZhang, 2019-05-24:
@EwoutH , it is difficult to explain it. gen9 HW didn't implement all VP9 features,  cant guarantee the quality and VP9 sub features enabling. another solution is hybrid solution( CPU & GPU work together to encode VP9) , but no much benefit comparing CPU solution.

Brainiarc7, 2019-06-11:
@EwoutH ,

Case in point: VP9 encoding on KBL is [limited to 60 fps only](https://github.com/intel/intel-vaapi-driver/pull/209), as the MFX block on that H/W gen is incomplete as far as VP9 encoding is concerned.
This is the kind of stuff a company such as Intel wouldn't want to declare prod ready in mediasdk and the media-driver projects, [considering the implications](https://github.com/intel/intel-vaapi-driver/issues/87).