intel/media-driver #634: Recoverable GPU hangs with multi-process h264 transcode

eero-t, 2019-05-28:
I'm seeing occasional recoverable GPU hangs with smaller resolution multi-process AVC transcoding.

Setup:
* HW: CFL GT3e (i5-8259U), KBL GT3e (i7-7567U), SKL GT3e (i5-6260U), 12 (J3455) or 18EU (J4205) BXT/APL, or BDW GT2 (i3-5010U)
* drm-tip git kernel v5.1 or newer
* git builds of the Media stack from within few weeks (libdrm, gmmlib, libva, media-driver, MediaSDK, ffmpeg, gst-msdk)

Test-cases:
* Do (say 17) parallel instances of small resolution AVC transcode with MSDK or FFmpeg QSV:
`sample_multi_transcode -i::h264 720x480p_30.00_4mb_h264_cabac_180s.264 -o::h264 output.h264 -b 2000 -u 4 -n 4800 -async 4 -hw`
`ffmpeg -hwaccel qsv -qsv_device /dev/dri/renderD128 -c:v h264_qsv -i 720x480p_30.00_4mb_h264_cabac_180s.264 -c:v h264_qsv -b:v 2000K -compression_level 4 -frames 4800 -y output.h264`

Expected outcome:
* No GPU hangs, recoverable or otherwise

Actual outcome:
* Every 10th to 20th round, there are recoverable GPU video engine hangs:
```
[12884.141579] i915 0000:00:02.0: GPU HANG: ecode 9:0:0x00000000, hang on vcs0
[12884.141587] [drm] GPU hangs can indicate a bug anywhere in the entire gfx stack, including userspace.
[12884.141588] [drm] Please file a _new_ bug report on bugs.freedesktop.org against DRI -> DRM/Intel
[12884.141590] [drm] drm/i915 developers can then reassign to the right component if it's not a kernel issue.
[12884.141592] [drm] The gpu crash dump is required to analyze gpu hangs, so please always attach it.
[12884.141593] [drm] GPU crash dump saved to /sys/class/drm/card0/error
[12884.141643] i915 0000:00:02.0: Resetting vcs0 for hang on vcs0
```

Notes:
* These hangs seem to have started about 3 weeks ago (around the same time kernel v5.1 got tagged and we updated it i.e. it could also be kernel related)
* Hangs seem to happen less rarely on KBL and SKL than on CFL GT3e & BXT devices
* I haven't seen them on GEN9 (SKL/KBL/CFL) GT2 devices, but that may be just luck with timings
* Sometimes this hang has happened also with FFmpeg VAAPI (on J3455), and other test-cases using VAAPI (on J4205), so it doesn't seem like a MSDK specific issue.  I assume FFmpeg VAAPI backend threading differences just happen to make this less likely to happen
* A lot of other media tests are run before these hangs happen.  I don't know whether any of those are required to trigger the hangs

eero-t, 2019-05-31:
J4205 BXT/APL devices hard hang now sometimes in following MSDK test:
`sample_multi_transcode -i::h264 1280x720p_29.97_10mb_h264_cabac.264 -o::h264 output.h264 -b 800 -u 4 -n 2400 -f 15 -w 352 -h 240 -FRC::PT -async 4 -hw`

eero-t, 2019-06-27:
> J4205 BXT/APL devices hard hang now sometimes in following MSDK test

BXT issue is not a hard hang or related to this issue, but a i915 driver deadlock without GPU hang: https://bugs.freedesktop.org/show_bug.cgi?id=110848

Regression here, i.e. random recoverable GPU hangs with multiple MSDK and FFmpeg instances running in parallel have continued with latest Git versions.  Here's error state for one of them:
https://bugs.freedesktop.org/attachment.cgi?id=144597

PS. How do I add attachments to these tickets?

eero-t, 2019-07-08:
These recoverable GPU hangs still continue.

> I haven't seen them on GEN9 GT2 devices, but that may be just luck with timings

They happen also on GT2 devices.  Currently I'm seeing them more there than on the single KBL GT3e that I'm still testing.

> Sometimes this hang has happened also with FFmpeg VAAPI (on J3455), and other test-cases using VAAPI (on J4205), so it doesn't seem like a MSDK specific issue. I assume FFmpeg VAAPI backend threading differences just happen to make this less likely to happen

I've seen them happen with VAAPI on KBL GT3e on too, not just BXT.   These recoverable hangs are much rarer with VAAPI though, than with MSDK or FFmpeg QSV.

wangyan42164, 2019-07-16:
@eero-t This issue happens on KBL GT2. Right? Could you share your video clip for debugging? Thanks.

eero-t, 2019-08-13:
I don't have anymore KBL GT2, only BDW & SKL GT2, KBL GT3e and BXT J4205.  From last one I'm not getting regular data due to: https://bugs.freedesktop.org/show_bug.cgi?id=110848

Of the above listed 3 Core devices, I've gotten Media GPU hang only once in last 4 weeks, it was yesterday, on KLB GT3e, in the same video file as earlier (hangs seem to have gone mostly away after upgrading to 5.2 drm-tip kernel).

I don't know whether I can provide the video publicly, but you can get it internally from the Intel MediaSDK validation team, its their SD03 test file. Because of the rarity of the hang occurance, you may have problem in reproducing it, but I can provide i915 error files, just tell me where.

eero-t, 2019-10-11:
> BXT issue is not a hard hang or related to this issue, but a i915 driver deadlock without GPU hang:
> https://bugs.freedesktop.org/show_bug.cgi?id=110848

Above kernel bug is now fixed, and I haven't seen Media GPU hangs for a while -> closing as fixed.

closed 2019-10-11 by eero-t