intel/media-driver #637: [regression] +378 vaapi conformance failures since c1e0c23bf05

uartie, 2019-06-04:
Since commit c1e0c23bf050815a38dc37301361d943e4c611f9, there are 378 new conformance test failures (test_va_api from https://github.com/intel/libva-utils):

```
commit c1e0c23bf050815a38dc37301361d943e4c611f9
Author: Dmitry Ermilov <dmitry.ermilov@intel.com>
Date:   Tue Apr 30 10:41:17 2019 -0400

    [Encode] Fix for race condition in AVC FEI
    
    This change aims to replace member FeiFunction of 'global' DDI_MEDIA_CONTEXT class
    with retrieving fei function type from provided vaConfig.
    Old implementation had a race condition bug when e.g. PREEnc and AVC FEI encode
    are created simultaneously from two threads.
    
    Fixes #590
    
    Signed-off-by: Dmitry Ermilov <dmitry.ermilov@intel.com>
```

...previously (prior to this commit) there were only 104 outstanding failures.

```
$  LIBVA_DRIVER_NAME=iHD test_va_api

[==========] Running 17415 tests from 14 test cases.
[----------] Global test environment set-up.
<snip>
[==========] 17415 tests from 14 test cases ran. (25145 ms total)
[  PASSED  ] 16933 tests.
<snip>
482 FAILED TESTS
```

uartie, 2019-06-05:
Yep, #638 fixes this regression for me ;)

dmitryermilov, 2019-06-06:
Thanks for confirmation :)

PR #638 (2019-06-11): Fixed HEVC FEI config creation
- Added CQP mode to a list of supported HEVC FEI VA configs
- Treat FEI mode as ENCPAK if no VAConfigAttribFEIFunctionType provided
- Fixed query of supported attributes for HEVC FEI

Fixes #637
commit 7aeb5356 (2019-06-11): Fixed HEVC FEI config creation

- Added CQP mode to a list of supported HEVC FEI VA configs
- Treat FEI mode as ENCPAK if no VAConfigAttribFEIFunctionType provided
- Fixed query of supported attributes for HEVC FEI

Fixes #637
closed 2019-06-11 by intel-mediadev