intel/media-driver #639: [ffmpeg-vaapi][transcode] transcode with scale_vappi failed

zhuqingliang, 2019-06-10:
cmdline:
ffmpeg -v verbose -init_hw_device vaapi=hw:/dev/dri/renderD128 -filter_hw_device hw -hwaccel_output_format vaapi -hwaccel vaapi -c:v h264 -i /root/media/testing/assets/avc/720x480p_30.00_4mb_h264_cabac_180s.264 -an -vf 'hwupload,scale_vaapi=w=4096:h=2304' -c:v hevc_vaapi -vframes 200 -y dump.265

error message:
[Parsed_scale_vaapi_0 @ 0x55b6337b6d40] Failed to render parameter buffer: 18 (invalid parameter).
Error while filtering: Input/output error
Failed to inject frame into filter network: Input/output error
Error while processing the decoded data for stream #0:0

wangyan-intel, 2019-06-10:
@zhuqingliang I tried some H264 videos on my hand but no this issue with the same command based on the latest media driver. If you have still this issue, could you please share your video stream? Thanks

zhuqingliang, 2019-06-14:
@wangyan-intel After update my ffmpeg environment, this issue can not reproduce, thank you for the information.

closed 2019-06-14 by zhuqingliang