intel/media-driver #640: [Gst-vaapi][VPP][deinterlace] Bob and Motion-adaptive got the same result.

feiwan1, 2019-06-10:
When set the deinterlace mode as Bob and Motion-adaptive, the output are bit match:

$ gst-launch-1.0 -vf filesrc location=1080p_MotionTest.yuv num-buffers=20 ! rawvideoparse format=nv12 width=1920 height=1080 interlaced=true top-field-first=1 ! vaapipostproc format=nv12 width=1920 height=1080 deinterlace-mode=1 deinterlace-method=bob ! checksumsink2 dump-output=true dump-location=./de.bob.nv12

$ gst-launch-1.0 -vf filesrc location=1080p_MotionTest.yuv num-buffers=20 ! rawvideoparse format=nv12 width=1920 height=1080 interlaced=true top-field-first=1 ! vaapipostproc format=nv12 width=1920 height=1080 deinterlace-mode=1 deinterlace-method=motion-adaptive ! checksumsink2 dump-output=true dump-location=./de.ma.nv12

$md5sum de*.nv12
1ebd131328ce1475f98f623fd11b34ae  de.bob.nv12
1ebd131328ce1475f98f623fd11b34ae  de.ma.nv12

Expected result:
dump output nv12 file de.bob.nv12 should not bit match with de.ma.nv12.

FurongZhang, 2019-06-13:
Adding Jason to check this issue.

FurongZhang, 2019-06-13:
Jason, is it possible for you to assign this issue to yourself? Sorry, I did not know your account.

Jister, 2019-06-17:
@feiwan1 We have kernel solution and Vebox solution for BOB DI, and media driver use VEBOX to do it for better quality. For motion adaptive, app need to give reference frame(previous frame) to driver, but gst-vaapi didn't give reference, thus result is same to BOB. The issue need gst-vaapi's change to pass reference frame to driver.

feiwan1, 2019-06-17:
@Jister , looks media-driver performs different behavior with vaapi-drvier. When using vaapi-driver, it doesn't have this issue. @xhaihao , do you know how vaapi-driver deal with Motion-adaptive deinterlace? Is that make sense to provide an additional reference frame in middle-ware ?

Jister, 2019-06-17:
@feiwan1 By the way, can you help to compare the results between media-driver and vaapi-driver, what's the value of md5sum of vaapi-driver using the same input?

feiwan1, 2019-06-18:
@Jister , here is vaapi-driver's result:
root@T:~/media/dein# md5sum de.*
1ebd131328ce1475f98f623fd11b34ae  de.bob.nv12
6aff6b7352dfbf757f4e4307b16dae8a  de.ma.nv12

And compared the 2 output files, there are lots of difference.

Jister, 2019-06-18:
@feiwan1 Thanks for the comparison, so the difference between media-driver and vaapi-driver is just the behavior when handling motion adaptive case. There is an example in Libva, https://github.com/intel/libva/blob/b8d1208a63e316e12e96c7916f40eee636fd555f/va/va_vpp.h#L223, here the reference frame is for motion adaptive deinterlace

yasij, 2020-01-10:
As I mentioned in #804, I checked mpv, kodi and gstreamer. All of them have code to pass reference frames in VAProcPipelineParameterBuffer.

However, the driver defines DDI_CODEC_NUM_BK_REF and DDI_CODEC_NUM_FWD_REF as 0. That's why the reference frames aren't passed. The driver reports that max number of FWD and BK references as 0 in DdiMedia_QueryVideoProcPipelineCaps.

See: [media_libva.h#L82](https://github.com/intel/media-driver/blob/master/media_driver/linux/common/ddi/media_libva.h#L82)

intel-vaapi-driver increments num_forward_references to 1 when reporting capabilities for madi and mcdi. If madi is working for this driver, it needs to give the correct num_backward_references and num_forward_reference in DdiMedia_QueryVideoProcPipelineCaps for that filter for clients to provide the reference frames.

Jister, 2020-05-29:
Hi @feiwan1 , could you please try #949

Jister, 2020-06-04:
Seems driver and gst-vaapi have different understanding about backward reference and forward reference. For deinterlace, driver regard past frame as backward reference, future frame as forward referece, but gst-vaapi regard past frame as forward reference, future frame as backward reference.

closed 2020-06-09 by intel-mediadev

yasij, 2021-03-15:
> Seems driver and gst-vaapi have different understanding about backward reference and forward reference. For deinterlace, driver regard past frame as backward reference, future frame as forward referece, but gst-vaapi regard past frame as forward reference, future frame as backward reference.

Are you sure it's not the driver that has this wrong? Have you compared it to the i965-va-driver implementation? I found [this discussion](https://mailman.videolan.org/pipermail/vlc-devel/2017-July/114119.html) on the VLC mailing list which states the opposite based on the FFmpeg implementation. That might explain why i965-va-driver is setting num_forward_references = 1, and num_backward_references = 0. It might also explain why deinterlacing was jerky when I enabled this patch.

PR #949 (2020-06-09): [VP] Correct reference number in caps
For advanced DI, the reported capability of backward reference number should be 1. Fix #640