intel/media-driver #643: [ffmpeg-vaapi][max_frame_size][vaGetConfigAttributes][Enhancement]

fulinjie, 2019-06-12:
Since VAConfigAttribMaxFrameSize has been added in libva,https://github.com/intel/libva/commit/dcbd65bf03d0755bbdaa1e69b81480e8de1fb8e2
IMHO, m_attributes for m_profileEntryTbl need to be added too to make the query for max_frame_size valid.

Message:
[h264_vaapi @ 0x5594b51cbec0] Max frame size attribute is not supported.

p (*m_profileEntryTbl[1].m_attributes)
$21 = std::map with 8 elements = {[VAConfigAttribRTFormat] = 131075, [VAConfigAttribEncryption] = 2147483648, [VAConfigAttribDecSliceMode] = 3, [VAConfigAttribDecProcessing] = 1, 
[VAConfigAttribMaxPictureWidth] = 8192, [VAConfigAttribMaxPictureHeight] = 8192, [VAConfigAttribProcessingRate] = 2, [VAConfigAttribCustomRoundingControl] = 1}

wangyan42164, 2019-06-19:
@fulinjie I submitted #656 to fix it. It works fine on my side. Could you please try it? Thanks.

fulinjie, 2019-06-19:
@wangyan42164 thanks Yan, it works in avc encoder and returns expected attrib.value for VAEncMiscParameterTypeMultiPassFrameSize.

A remained concern:
Is there any plan to support VAEncMiscParameterTypeMaxFrameSize which is also defined in libva?

If no, i would like to consider the multi path solution only.

wangyan42164, 2019-06-19:
@fulinjie VAEncMiscParameterTypeMaxFrameSize has been supported in current media driver.

wangyan-intel, 2019-06-24:
@fulinjie I checked the code of media driver. VAEncMiscParameterTypeMaxFrameSize should be for VP8/9 only.  VAEncMiscParameterTypeMultiPassFrameSize is for AVC only. Thanks.

fulinjie, 2019-06-24:
Thanks a lot.

PR #656 (2019-07-12): Add VAConfigAttribMaxFrameSize query support for AVC encoding.
Fixes #643.
commit 649e8595 (2019-07-12): Add VAConfigAttribMaxFrameSize query support for AVC encoding.

Fixes #643.
closed 2019-07-12 by intel-mediadev