intel/media-driver #644: Gstreamer can not convert the ARGB color to NV12 correctly.

HeJunyan, 2019-06-12:
The gstreamer pipeline 

./build/bin/gst-launch-1.0 videotestsrc ! capsfilter caps=video/x-raw,format=ARGB ! vaapipostproc ! video/x-raw,format=NV12 ! vaapisink

get wrong result

while 
./build/bin/gst-launch-1.0 videotestsrc ! capsfilter caps=video/x-raw,format=RGBA ! vaapipostproc ! video/x-raw,format=NV12 ! vaapisink
and
./build/bin/gst-launch-1.0 videotestsrc ! capsfilter caps=video/x-raw,format=BGRA ! vaapipostproc ! video/x-raw,format=NV12 ! vaapisink

works OK.

Notice 
const VAImageFormat m_supportedImageformatsG11[] =
{   {VA_FOURCC_BGRA,   VA_LSB_FIRST,   32, 24, 0x00ff0000, 0x0000ff00, 0x000000ff,  0xff000000},
    {VA_FOURCC_ARGB,   VA_LSB_FIRST,   32, 24, 0x00ff0000, 0x0000ff00, 0x000000ff,  0xff000000},

BGRA and ARGB have the same RGB channel mask,
may be wrong

FurongZhang, 2019-06-17:
Yes, That is a bug.

FurongZhang, 2019-06-17:
https://github.com/intel/media-driver/pull/653  I am verifying this changelist. Will let you know when I finish the validation and check if this code changes can fix this issue or not. Thanks in advance!

FurongZhang, 2019-07-04:
I have checked in the code, is it possible for you to verify and close this issue? Please let me know if you still have any issues. Thanks a lot in advance! @HeJunyan

HeJunyan, 2019-07-08:
After my debugging, it seems that the forcc ARGB and BGRA map to the same image format, the channels order is same for these two formats. Please help to check, thanks

FocusLuo, 2019-12-05:
@FurongZhang Any update on this issue?

MicroYY, 2022-01-18:
Tried Furong's fix #653. Still now work. Debugging WIP.

wennibao, 2024-08-09:
duplicate to https://github.com/intel/media-driver/issues/1796

wennibao, 2024-08-12:
Fixes: intel/media-driver#644