intel/media-driver #662: [gstreamer-vaapi][jpeg decode] fail to decode 4:0:0 chroma type jpeg into Y800 format.

feiwan1, 2019-06-25:
Hi,
Now we are using gst-vaapi to decode a jpeg with Y channel only(grey jpeg), ideally it should be decode into Y800 format automatically, but now some va api doesn't support Y800 in media-driver, so now it will be decoded into NV12 in gst-vaapi side. So could help to support VA_FOURCC_Y800 into below apis?
1. vaQueryImageFormats
2. vaCreateImage
3. vaGetImage

jpeg info by mediainfo:
root@T:~# mediainfo smpte_Y800.jpeg 
General
Complete name                            : smpte_Y800.jpeg
Format                                   : JPEG
File size                                : 33.2 KiB
Image
Format                                   : JPEG
Width                                    : 720 pixels
Height                                   : 480 pixels
Color space                              : Y
Bit depth                                : 8 bits
Compression mode                         : Lossy
Stream size                              : 33.2 KiB (100%)

gstreamer cmdline:
gst-launch-1.0 filesrc location=smpte_Y800.jpeg ! jpegparse ! vaapijpegdec ! filesink location=out.yuv

gstreamer-vaapi repo:
https://gitlab.freedesktop.org/wangfei/gstreamer-vaapi branch: jpeg_gray8
[smpte_Y800.jpeg.tar.gz](https://github.com/intel/media-driver/files/3324251/smpte_Y800.jpeg.tar.gz)

feiwan1, 2019-06-26:
BTW, if can add 411 chroma format(411P), that will be great.
411 jpeg file:
[411.jpeg.tar.gz](https://github.com/intel/media-driver/files/3328628/411.jpeg.tar.gz)

Jister, 2019-07-01:
@feiwan1 Could you please try #669 ?

Jister, 2019-07-15:
@feiwan1 have you verified the patch? If it works fine, will start to merge the PR#669

feiwan1, 2019-07-15:
HI @Jister , sorry for missing your last ping, you patch looks good for me, some error info need to be removed from gst-vaapi side. Please help to merge your parch. And thanks for your great support.

feiwan1, 2019-09-06:
Close this issue, since #669 merged. Thanks.

closed 2019-09-06 by feiwan1