intel/media-driver #664: unsupported VA image format nv12 with Iris mesa driver (19.1.x)

enihcam, 2019-06-26:
Repro steps:

mesa 19.1.1 with `MESA_LOADER_DRIVER_OVERRIDE=iris`
intel-media-driver 19.1.0 with `LIBVA_DRIVER_NAME=iHD`

mpv with an H264 video, and you shall see no pictures at all, with logs:
```
Using hardware decoding (vaapi).
AO: [pulse] 48000Hz stereo 2ch float
VO: [gpu] 1280x534 => 1281x534 vaapi[nv12]
[vo/gpu/vaapi-egl] unsupported VA image format nv12
[vo/gpu] Initializing texture for hardware decoding failed.
```

wangyan-intel, 2019-06-27:
@enihcam Could you share your video clips? And what OS you run MPV? Ubuntu? Fedora? Thanks.

enihcam, 2019-06-27:
@wangyan-intel Unfortunately I cannot share my video clip.

The OS is Archlinux, with all packages sync-ed. Here is the mpv.conf:

```
vo=gpu
hwdec=vaapi
```

FurongZhang, 2019-07-04:
I saw yan had followed up this issue. So assign it to Yan. Thanks Yan in advance! Maybe related to video decoding.

enihcam, 2019-07-07:
Related issue:
https://bugzilla.freedesktop.org/show_bug.cgi?id=110666

wangyan-intel, 2020-03-11:
Based on https://bugzilla.freedesktop.org/show_bug.cgi?id=110666, 

Kenneth Graunke 2019-05-17 02:47:16 UTC
That makes sense.  Commit 4b5e8eb3c8d709bd7c6d1a33a114bf4b002548f8 from today should also help with NV12 support.

We're not backporting iris fixes to 19.1; feel free to use master.

closed 2020-03-11 by wangyan-intel