intel/media-driver #671: Endurance test's fails due to polling in driver

alariono, 2019-07-02:
This issue is very related to https://github.com/intel/media-driver/issues/343 
Endurance test’s fails after updating kernel [4.9.5]->[5.0.0].
Linux kernel’s patch: https://cgit.freedesktop.org/drm/drm/commit/?id=b16c765122f987056e1dc9ef6c214571bb5bd694 
Feedback from kernel’s team: “Proper fix on userspace side is not to poll with sleeps, especially not short ones”
But proper fix in UMD is actually https://github.com/intel/media-driver/issues/343 request.

Steps for reproduce:
1. Build full media stack: MediaSDK, UMD driver and etc.
2. Run SMT for transcoding with attached par file lstN15N-all_skl.par.
“sample_multi_transcode -par lstN15N-all_skl.par -timeout 90000”
3. In separate console run extra SMT for transcoding. This one transcoding session will fail in 5-10 seconds.
“sample_multi_transcode -i::h264 1920x1088_frm_cab_24mbs.264 -o::mpeg2 1920x1080_2400.mpeg2 -hw -async 1 -b 9600”

[lstN15N-all_skl.par.txt](https://github.com/intel/media-driver/files/3350591/lstN15N-all_skl.par.txt)

alariono, 2019-07-02:
+ @dmitryermilov, @onabiull, @dvrogozh

dmitryermilov, 2019-07-02:
+ @Xiaogang-Li , @Xiaogangli-intel

alariono, 2019-07-03:
+@alariono

XinfengZhang, 2019-07-05:
I wonder why the issue is related with the frequent sleep. the code is very old, why KMD promotion will impact it?

alariono, 2019-07-05:
Hello XinfengZhang,
I will forward discussion to you

Xiaogangli-intel, 2019-09-04:
does this issue still here when update the KMD?

Xiaogangli-intel, 2020-03-24:
The polling way in driver is a long story, should not cause this issue. From the description, I think this issue should be caused by KMD updating.

Maybe this issue should be fixed by the latest KMD, will close this issue if no other concern.

Link the answer for #343,
https://github.com/intel/media-driver/issues/343#issuecomment-603245047

closed 2020-03-26 by Xiaogangli-intel