intel/media-driver #672: RGB16(565) generate garbage in postproc for Gstreamer

HeJunyan, 2019-07-10:
The command line
gst-launch-1.0 videotestsrc ! capsfilter caps=video/x-raw,format=RGB16,width=800,height=600 ! vaapipostproc ! video/x-raw,format=NV12 ! vaapisink

generate garbage in skylake
while the comment id:

commit 296c5fb61c812be2e82ae9857ac15c8c160561ec
Author: Chen, Jason K <jason.k.chen@intel.com>
Date:   Fri Mar 22 09:34:31 2019 +0800

    [VP]Add RGB565 support in vaCreateImage

Works well.

wangyan-intel, 2019-07-11:
@Jister Could you help check it? Thanks.

Jister, 2019-07-11:
@HeJunyan, as I dump the input and output data, seems input data is not correct, could you help to check whether gstreamer pass correct input data to driver?
[dumpData.zip](https://github.com/intel/media-driver/files/3381144/dumpData.zip)

HeJunyan, 2019-07-11:
But i check
commit 296c5fb
can really work well, what is the difference?

HeJunyan, 2019-07-11:
@Jister
I tried all white input using
gst-launch-1.0 videotestsrc pattern=white ! capsfilter caps=video/x-raw,format=RGB16,width=800,height=600 ! vaapipostproc ! video/x-raw,format=NV12 ! vaapisink

And all the input data is a 0xffff when I dump input, but still see garbage lines.
I see a lot of dither related modification include RGB16 format, is that have some effect?

Jister, 2019-07-12:
Wrong input data is caused by wrong tile format between image and surface in PutImage. Will provide a fix for it.

Jister, 2019-07-12:
@HeJunyan please try #680

Jister, 2019-07-19:
@HeJunyan have you verified the patch? If it works fine, will start to merge the PR#680

HeJunyan, 2019-07-19:
Verified and OK, thanks

PR #680 (2019-07-19): [VP] Add tile type for image creation
Fix #672
closed 2019-07-19 by intel-mediadev